dhalperi/linux-80211n-csitool-supplementary #2: figure out how to add dpward to list of authors and/or other credits

dhalperi, 2015-02-03:
@dpward  -- however you want to be acknowledged as part of this effort is up to you. Feel free to change or amend the web page as you see fit. Thanks!

chenyongxi, 2015-02-04:
Excuse me，can we use this collect CSI from multiple APs at the same time?    
Looking forward to your answer！

dhalperi, 2015-02-04:
@chenyongxi this is a completely inappropriate way to engage. This issue has a purpose and your problem is unrelated to it.

Please read the [instructions](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/CONTRIBUTING.md) about how to get help.

Your question is answered in the FAQ.

dpward, 2015-02-04:
Updated "Credits" section on the main page of the website.

closed 2015-02-04 by dpward