dhalperi/linux-80211n-csitool-supplementary #3: can't read private key when Installing the modified driver in step 2

jinzhang-unsw, 2015-02-05:
I copy and paste the steps and always have the can't read private key information, after I type the cmd:
sudo make -C /lib/modules/$(uname -r)/build M=$(pwd)/drivers/net/wireless/iwlwifi INSTALL_MOD_DIR=updates \
    modules_install

And step 4 has errors about missing the link files, which could be the cause of the last problem.
I use the linux 14.04. Kernel version is 3.13.0-32-generic #57-Ubuntu.
Do you know how to fix that?

Jin

dpward, 2015-02-05:
The private key that Ubuntu uses to sign kernel modules is not distributed, which is why you are seeing the warning, but you should be able to safely it. (I'll update the installation instructions.) If there's an actual problem arising from that, you'll see it in step 5.

Could you please paste the specific error you are seeing in step 4? It looks like you're missing a development package. Thanks!

jinzhang-unsw, 2015-02-05:
Yeah maybe. The following are the part of the error info. Thanks!
In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:38:0,
                 from /usr/include/netinet/in.h:23,
                 from /usr/include/arpa/inet.h:22,
                 from log_to_file.c:4:
/usr/include/x86_64-linux-gnu/bits/socket.h:213:0: note: this is the location of the previous definition
 #define MSG_FASTOPEN MSG_FASTOPEN
 ^
In file included from log_to_file.c:6:0:
/lib/modules/3.13.0-32-generic/build/include/linux/socket.h:259:0: error: "MSG_CMSG_CLOEXEC" redefined [-Werror]
 #define MSG_CMSG_CLOEXEC 0x40000000 /* Set close_on_exit for file
 ^
In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:38:0,
                 from /usr/include/netinet/in.h:23,
                 from /usr/include/arpa/inet.h:22,
                 from log_to_file.c:4:
/usr/include/x86_64-linux-gnu/bits/socket.h:218:0: note: this is the location of the previous definition
 #define MSG_CMSG_CLOEXEC MSG_CMSG_CLOEXEC
 ^
In file included from /lib/modules/3.13.0-32-generic/build/include/linux/kernel.h:6:0,
                 from /lib/modules/3.13.0-32-generic/build/include/linux/skbuff.h:17,
                 from /lib/modules/3.13.0-32-generic/build/include/linux/netlink.h:6,
                 from log_to_file.c:7:
/lib/modules/3.13.0-32-generic/build/include/linux/linkage.h:7:25: fatal error: asm/linkage.h: No such file or directory
 #include <asm/linkage.h>
                         ^
cc1: all warnings being treated as errors
compilation terminated.

dhalperi, 2015-02-05:
Hi @jinzhang-unsw -- please enclose copied-and-pasted logs in three backticks: ```

This will make the code be readable on github. Please see how I edited your comment above to make this work :)

jinzhang-unsw, 2015-02-05:
Ok I see.
```
thank you ;)
```

dpward, 2015-02-06:
@jinzhang-unsw Please pull the updates I just added to the linux-80211n-csitool-supplementary repository and try again.

I believe this error is caused by including kernel headers from userspace. That works fine for me and was used in the past with this code, but [it is discouraged](http://lwn.net/Articles/113349/), so I changed the Makefile to only include the C library headers.

For Ubuntu 14.10 this should be all you need to do. (However, in LTS releases of Ubuntu where the major version of the kernel has been updated from the initial release, it is possible that the value of CN_NETLINK_USERS in <linux/connector.h> might be different between the C library headers and kernel source headers. In that case, the C library headers need to be edited to reflect this change. I'll add a step to the installation for verifying and, if necessary, editing this value to match.)

Please let me know if this solves your issue.

jinzhang-unsw, 2015-02-06:
After I excute step 4 again, it shows a series of gcc  -Wall -Werror *.o ... info. I think it works.
After excuting step 5
```
sudo modprobe -q -r iwldvm iwlwifi mac80211
```
the result is 
```
rmmod: ERROR: missing module name.
```
After the other one command, the wifi looks work and I can use iw to scan the AP. I am not sure if the intermediate results have problems.

dpward, 2015-02-06:
@jinzhang-unsw This error can be safely ignored as well.

Ubuntu includes a configuration file /etc/modprobe.d/iwlwifi.conf, which tries to run ```rmmod iwldvm``` when the ```iwlwifi``` module is removed. Since the ```iwldvm``` module was already removed though, the error you mentioned appears. Unfortunately, this configuration file doesn't exist on other distributions besides Ubuntu: if ```iwldvm``` is not unloaded first then ```iwlwifi``` will fail to unload.

I will also update the installation instructions to this effect. However you should have everything set up correctly now.

closed 2015-02-06 by dpward

jcdhlzq, 2017-02-25:
How did you solve the problem "can't read private key" ? I have tried to set enforcemodulesig=0 and restarted. It did not work as well. The problem still exists.

jcdhlzq, 2017-02-25:
How did you solve the problem "can't read private key" ? I have tried to set enforcemodulesig=0 and restarted. It did not work as well. The problem still exists. Thank you!

nafiul-nipu, 2017-05-27:
i am having the same problem..  can you tell me how you solved it