dhalperi/linux-80211n-csitool-supplementary #5: How to restore the unmodified driver?

linkin8834, 2015-02-24:
Hello,
    I am using Ubuntu 14.04 with a working WiFi LINK 5300. I backup the old driver using 

for file in /lib/firmware/iwlwifi-5000-*.ucode; do sudo mv $file $file.orig; done

 I am using iwlwifi-5000-5.ucode. After experiment, I try to switch back to the old working driver by 

cp iwlwifi-5000-5.ucode.orig iwlwifi-5000-5.ucode

After restart the network manager, the WiFi does not work. I am having deauthenization problem all the time (with reason 3).

  I notice that there is also iwlwifi-5000.2.ucode due to 

sudo ln -s iwlwifi-5000-2.ucode.sigcomm2010 /lib/firmware/iwlwifi-5000-2.ucode

 So I change the name to iwlwifi-5000-2.ucode. After restart the network-manager, still I cannot connect to any WiFi.

 Is there any procedures that I should follow to 'restore' the old working intel WiFi 5000 series driver? Thanks!

linkin8834, 2015-02-25:
Update:
   I am getting deauthenticating from xxxx by local choice (reason=3) all the time after I use the modified firmware.

linkin8834, 2015-02-25:
Update:
    I can connect to WiFi now. However, everytime I run 
sudo modprobe -r iwlwifi mac80211
and then 
sudo modprobe iwlwifi connector_log=0x1

I cannot connect to ANY open WiFi. Thus, I cannot do any measurement.

I set up a wireless router without encryption for testing purpose. But still I cannot connect to it.

dpward, 2015-02-25:
Hi!

In addition to restoring the original firmware, you need to remove the modified driver itself. To do this, run:
```
sudo rm -f /lib/modules/$(uname -r)/updates/{iwlwifi,iwldvm,iwlmvm}.ko
sudo depmod
```

Also about the steps you took to restore the firmware: you will actually want to change the filename of the modified firmware back to `iwlwifi-5000-2.ucode.sigcomm2010`, or just delete it. The driver will only try to load firmware named `iwlwifi-5000-#.ucode`, including symlinks that have this name that point to another file, so we want to make sure it won't l

Please let me know if this solves your problem.

linkin8834, 2015-02-25:
Hello,
    Thanks for your reply! I will try that later today. 
    
    I managed to connect to my open WiFi after I disable the wireless interface according to FAQ. It seems that the network manager keeps me from connecting to any WiFi after modprobe the iwlwifi with connector_log=0x1. 

    I run the 'log tool' in one terminal, and ping my wifi router in another terminal. However, I see nothing in the first terminal, nor anything in csi.dat.

   The router has support for 802.11n. 
    
   I have only one WiFi 5300 wifi card. Is there any suggestions on the access point? Should I get another 5300 or just use wireless router for experiment?
    
  Thanks!

dpward, 2015-02-25:
Are you sure the custom firmware is loaded? In your first comment, you had restored the original firmware &mdash; did you change it back?

linkin8834, 2015-02-25:
I think I have re-run
sudo ln -s iwlwifi-5000-2.ucode.sigcomm2010 /lib/firmware/iwlwifi-5000-2.ucode
so the driver should be the modified one.

dpward, 2015-02-25:
Okay, but does `iwlwifi-5000-5.ucode` still exist? If so, it will get loaded instead of the modified firmware. It needs to be renamed to `iwlwifi-5000-5.ucode.orig`.

linkin8834, 2015-02-25:
I think that is the issue. I still keep the -5.ucode in /lib/firmware/ since I was thinking that -2.ucode would be loaded on system boot by default...
Let me remove the -5.ucode and try again tomorrow.

Thanks!

linkin8834, 2015-02-26:
Hello,
    Today I remove the -5.ucode firmware, and re-do the measurement by setting connector_log to 0x5. I can see a lot of messages in terminal like:
    received 233 bytes: id: 26 val :1 seq: 0 clen: 233

    And I ping the wireless router in another terminal. However when I try to load the CSI.dat using the matlab code read_bf_file, it always return an empty cell.
    Is there any suggestion on the wireless router? Should I just use another 5300 to communicate with the current one, so as to capture CSI?

   Thanks!

dpward, 2015-02-26:
The problem could be with how MATLAB is parsing the data, rather than how you are collecting it. Since you are logging both beamforming notifications and received payloads, you want to check to see if your `csi.dat` file contains any beamforming events in order to narrow down the problem first. Run this command:

```
linux-80211n-csitool-supplementary/netlink/parse_log csi.dat
```

Do you have any entries with `code=0xBB`? Those are beamforming notifications (i.e. CSI).

dhalperi, 2015-02-26:
I'm inclined to think that you are only logging packet payloads (0x4) and
not seeing any beamforming entries (0x1 in the connector_log mask).

linkin8834, 2015-02-26:
Hello,
       I use parse_log to see my csi.dat, and I have the following:
       Entry size=233, code=0xC1
       I search through the file, and there is no 0xBB (beamforming) packages.
       I am only using two antennas out of three antennas of the Intel WiFi 5300. 
       I set up a wireless router as access point. I am using ASUS RT-N56U dual-band wireless-N600 router. It supports 802.11n mode. I can connect to it, and could ping through PC using 5300 on PC.
      Thanks!

linkin8834, 2015-02-26:
Hello,
     I found from the SIGCOMM paper that the CSI subcarrier index is from -28 to 28. But the example log (log.all_csi.6.7.6) gives me 1x3x30 complex channels. Is there any function that I could use to get the frequency response from -28 to 28? Thanks!

dhalperi, 2015-02-26:
Please read #5 on the FAQ.
http://dhalperi.github.io/linux-80211n-csitool/faq.html

If the AP does not send packets using 802.11n rates, you will never see
recorded CSI measurements. Either you have not configured the AP and/or
client to use 802.11n rates, or you're not sending enough data to trigger
the rate-selection algorithm to switch to them, or something else is broken
in your configuration.

Dan

dhalperi, 2015-02-26:
Please see the FAQ, part 2B.
http://dhalperi.github.io/linux-80211n-csitool/faq.html

Dan

linkin8834, 2015-02-26:
Thanks! I will do more experiments.

linkin8834, 2015-02-26:
Hello,
     I did more experiments, but still cannot get beamforming packages. Just want to confirm, is it OK to use a WiFi router with 802.11n as the access point? Otherwise I'd better get another 5300 to pair with the current one to run the injection/monitor scripts. Thanks!

dhalperi, 2015-02-26:
As I said, if the AP is not using 802.11n rates, then there will be no CSI reports measured. Something must be wrong in your configuration.

linkin8834, 2015-02-27:
Hello,
    The problem is solved when I change my AP to another 802.11n wireless router and re-install the modified firmware. I have the CSI now. Thanks for your help!

closed 2015-02-27 by linkin8834

dhalperi, 2015-02-27:
great!

lidongmeng, 2015-06-19:
@linkin8834 hello, I caught the same problem as u. I use the same OS 14.04, and I do as the guide. But after I unload the driver and reload it with CSI logging enabled, I can not connect to any AP with no password, how do u deal with the problem, thanks!

linkin8834, 2015-06-19:
Hi Li,
     With CSI logging turned on and load the modified firmware, you cannot
use encrypted AP.

     The thing I do is I wrote a simple script to switch between 'Internet'
mode and 'Experiment' mode.

This is the content for change_to_5300_mode.sh:

#!/bin/bash/
if [ -f /lib/firmware/iwlwifi-5000-5.ucode ]; then
  mkdir -p ~/src/iwlwifi_backup
  mv /lib/firmware/iwlwifi-5000-5.ucode ~/src/iwlwifi_backup
fi

cp interfaces_manual_wireless /etc/network/interfaces
reboot

This is the content for connect_to_wifi.sh:

#!/bin/bash/
if [ -f /lib/firmware/iwlwifi-5000-5.ucode ]; then
mv /lib/firmware/iwlwifi-5000-5.ucode ~/src/iwlwifi_backup
cp /lib/firmware/iwlwifi-5000-5.ucode.orig
/lib/firmware/iwlwifi-5000-5.ucode
else
cp /lib/firmware/iwlwifi-5000-5.ucode.orig
/lib/firmware/iwlwifi-5000-5.ucode
fi
#modprobe -r iwlwifi mac80211
#sleep 1
#modprobe iwlwifi mac80211
cp interfaces_wireless /etc/network/interfaces
#service network-manager restart
reboot

This is the content for interfaces_wireless:

auto lo
iface lo inet loopback

This is the content for interfaces_manual_wireless:

auto lo
iface lo inet loopback
iface wlan0 inet manual

So, when you want to use internet, you use
bash connect_to_wifi.sh

and when you want to do experiment using the modified firmware, you use
bash change_to_5300_mode.sh

y119777, 2015-06-19:
i just did as you said above,the situation is just as following:
/etc/network/interface is:
auto lo
iface lo inet loopback
iface wlan0 inet manual
/lib/firmware file contained '5000' is following:
iwlwifi-5000-2-ucode
iwlwifi-5000-2-ucode.sigcomm2010
there is nothing contained '5000-5'

then i cant connect any AP,i hope u can help me

linkin8834, 2015-06-20:
Hello,
         Then you should restore the original code of iwlwifi-5000-2-ucode.
It should be in the folder /lib/firmware and named as
         iwlwifi-5000-2-ucode.orig
         You could modify my script, and replace '5' with '2'.

zjllynn, 2016-06-22:
Hello @linkin8834 ,

I have the same problem you have met. Once I run the sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat, and ping the router, I could not see and record any csi data which means nothing comes up in  terminal and the csi.dat is empty. How do you solve this problem? 
I used your script and it works well for my laptop and modified kernel are uploaded successfully. The AP is a router supporting 802.11n.

Thanks.

tahmidzbr, 2016-10-17:
@zjllynn Hi, did you solve the problem? if so can you please share here?

h11128, 2020-02-20:
After I do
```
sudo rm /lib/firmware/iwlwifi-5000-2.ucode
for file in /lib/firmware/iwlwifi-5000-*.ucode.orig; do sudo mv ${file} ${file%.orig}; done
sudo rm -f /lib/modules/$(uname -r)/updates/{iwlwifi,iwldvm,iwlmvm}.ko
sudo depmod
```
to restore the driver and firmware, and reboot my computer. The iwlwifi driver can not load normally anymore. It has several disagreement symbols with iwldvm module. I think the command above is not enough to restore the driver and firmware