dhalperi/linux-80211n-csitool-supplementary #6: why AOA is not stable?

xu-wang11, 2015-03-04:
I use csi tools to calculate the direction of angle. I draw a pseudospectrum using music algorithm. However, the peaks of pseudospectrums seems clusters to four group. I have no idea with this issue. Anyone know about it?

dhalperi, 2015-03-04:
The CSI on these devices may not be well-calibrated enough for certain uses... Intel did not actually ship any features that exploit the CSI in their hardware and there are a number of strange, lingering issues with it.

On 2.4 GHz bands, it appears that each antenna experiences an unknown phase shift by a multiple of π/2 (or 90 degrees). This is probably what you're seeing?

If I recall correctly, this effect does not show up at 5 GHz(*). So if you do your experiments in that band, you may not have this problem.

(*) Manikanta Kotaru from Stanford pointed out that this effect disappears on 5 GHz, and pointed at Section 3 in [Kyle Jamieson's ArrayPhaser](http://www.cs.ucl.ac.uk/fileadmin/UCL-CS/research/Research_Notes/ArrayPhaser_Enabling_Signal_Processing_on_WiFi_Access_Points_RN1404.pdf) paper as evidence.

Hope that helps!

dhalperi, 2015-03-04:
FYI @dpward – I didn't quite follow the changes you made to the FCC code, but I think your version of the CSI Tool works in fewer channels than the old stable version?

closed 2015-03-04 by dhalperi

dpward, 2015-03-04:
@dhalperi &mdash; in the old stable version, the driver just ignored any restrictions on 5 GHz channels that it read from the Intel adapter's EEPROM, and then it reported to Linux that the channels were unrestricted. This meant that Linux would allow the wireless interface to be configured in AP mode (for example) which it wasn't allowing before. However, that didn't make the adapter fully *work* unrestricted on 5 GHz channels &mdash; the firmware also reads the EEPROM and can block commands from the driver that do not meet its regulatory settings. If this happened, it was not at all apparent to the user what was wrong, because everything in Linux would show that the channel was unrestricted. (I'm not sure to what degree this impacted AP mode, but it completely prevented association from happening in IBSS mode.)

The better solution is to [fix the data in the adapter's EEPROM](https://code.google.com/p/iwleeprom/) to match the local regulatory settings, so that both the driver and firmware will work in beaconing modes (AP, IBSS, etc.) on channels where it is allowed. Then the driver can still indicate to the user if something is not allowed because of regulatory settings (for example, on the 5 GHz channels where DFS is required, the adapter should not be using these modes).

gsongsong, 2016-06-17:
I am sorry if this is not much related with Intel 5300 CSI Tool. But I found this sentences seem something related with my problem.

>On 2.4 GHz bands, it appears that each antenna experiences an unknown phase shift by a multiple of π/2 (or 90 degrees). This is probably what you're seeing?

> If I recall correctly, this effect does not show up at 5 GHz(*). So if you do your experiments in that band, you may not have this problem.

> (*) Manikanta Kotaru from Stanford pointed out that this effect disappears on 5 GHz, and pointed at Section 3 in Jon Gjengset, Graeme McPhillips, and Kyle Jamieson's ArrayPhaser (@jonhoo's) paper as evidence.

Does this apply to the Atheros' 802.11n chipse? i.e., AR9xxx series, chipsets operated by  `ath9k` driver. I mean I wonder whether unknown phase shift at 2.4 GHz is band-specific and manufacturer-invariant. Or is it just Intel-specific problem?

Currently, I am having difficulties on manipulating (estimated) true phase of CSI at 2.4 GHz bands.

saltdou, 2017-12-13:
@dhalperi  Have you solved the unexpected phase shift of pi/2 at 2.4 GHz? I also found a phase jump at 5 GHz, but it changes very slowly.

gsongsong, 2017-12-13:
Judging without a figure, that phase jump probably be due to phase wrapping.

saltdou, 2017-12-14:
No, the phase has be unwrapped. You can see my plot that is the relative
phase between antennas.
[image: Inline image 1]

gsongsong, 2017-12-14:
@saltdou Maybe I cannot solve the issue, but could you post a figure once again? It seemst attachment didn't work properly.

saltdou, 2017-12-14:
Yeah. I posted my measured results to you. It shows a phase shift of
multiple pi/2. I also found the similar phenomenon at 5GHz, but it shows
about every several mins rather than very quickly.