dhalperi/linux-80211n-csitool-supplementary #8: How can I get the CSI information from multiple AP at the same time?

chenyongxi, 2015-03-12:
Now, I can get the CSI information from the AP which I connected. However, I want get the CSI information from multiple AP at the same time. I see your FAQ answer 11. I did as he said. But I don't understand how to use it.   Who is receiver? Who is transmitter? How to use?

chenyongxi, 2015-03-12:
Look forward to your reply, very grateful！

dhalperi, 2015-03-12:
Say there are 5 devices in the network, A B C D and E. If you want to measure the CSI for links A->E, B->E, C->E, and D->E all at the same time, then ABCD are the transmitters and E is the receiver.

chenyongxi, 2015-03-13:
See your reply I am very happy. Thankyou! But I did not completely understand. what are A,B,C,D? AP or notebook? if A,B,C,D are AP, how to operate them ？

dhalperi, 2015-03-13:
A,B,C,D are transmitters from which you can inject packets to the broadcast address.

There is no way to listen to multiple commercial APs at once, as far as I know. Please read the FAQ again to understand why.

chenyongxi, 2015-03-13:
Are you mean no way to get the CSI information from multiple APs at the same time?

dhalperi, 2015-03-13:
Correct.

chenyongxi, 2015-03-13:
Thank you very much! If I have a problem I have to ask you to come back.

closed 2015-03-13 by dhalperi

chenyongxi, 2015-03-13:
Excuse me,if I use injection mode,can I get the CSI information from multiple APs at the same time?