dhalperi/linux-80211n-csitool-supplementary #9: Cannot compile the injection scripts

linkin8834, 2015-03-12:
Hello,
     I have svn'ed the lorcon-old, and build successfully on my Ubuntu 14.04. Here is what I have after make install of lorcon-old:

install -d -m 755 /usr/local/include
install -d -m 755 /usr/local/include/lorcon2/
install -m 644 lorcon.h /usr/local/include/lorcon2/lorcon.h
install -m 644 lorcon_packet.h /usr/local/include/lorcon2/lorcon_packet.h
install -m 644 lorcon_packasm.h /usr/local/include/lorcon2/lorcon_packasm.h
install -m 644 lorcon_forge.h /usr/local/include/lorcon2/lorcon_forge.h
install -m 644 ieee80211.h /usr/local/include/lorcon2/lorcon_ieee80211.h
install -d -m 755 /man/man3
install -o root -m 644 lorcon.3 /man/man3/lorcon.3

However, when I am trying to make install the injection code, I am getting

gcc -O3 -Wall -static -Wno-unknown-pragmas -I /usr/src/linux-headers-3.16.0-30-generic/include   -c -o util.o util.c
In file included from /usr/src/linux-headers-3.16.0-30-generic/include/linux/kernel.h:6:0,
                 from /usr/src/linux-headers-3.16.0-30-generic/include/linux/skbuff.h:17,
                 from /usr/src/linux-headers-3.16.0-30-generic/include/linux/if_ether.h:23,
                 from util.c:5:
/usr/src/linux-headers-3.16.0-30-generic/include/linux/linkage.h:7:25: fatal error: asm/linkage.h: No such file or directory
 #include <asm/linkage.h>
                         ^
compilation terminated.
make: *** [util.o] Error 1

I search the path. I have asm-generic in the folder 

/usr/src/linux-headers-3.16.0-30-generic/include

But I don't have the folder 'asm'.

Thanks!

dhalperi, 2015-03-12:
@dpward maybe this is also related to the issue you mentioned elsewhere about the perils of linking kernel headers from userspace code?

dpward, 2015-03-13:
@dhalperi Right, this Makefile shouldn't be including kernel source headers either. Fixed in 1fbb7f0820f9601fc40cad2499c8faa9e5916746.

@linkin8834 Please run `git pull` to obtain the fix to the injection code and then try again. Thanks for letting us know!

closed 2015-03-13 by dpward