dhalperi/linux-80211n-csitool-supplementary #10: Can I can I get the CSI information from multiple APs at the same time if I use injection mode

chenyongxi, 2015-03-13:
Can I can I get the CSI information from multiple APs at the same time if I use injection mode? What are the advantages of the injection mode ？

dhalperi, 2015-03-13:
If you use injection mode, the receiver can get CSI from all the transmitters. Transmitter is a different thing than AP, however.

chenyongxi, 2015-03-13:
thankyou! what is transmitter in the injection mode?

dhalperi, 2015-03-13:
the device you run the transmit code on. Did you look at the injection scripts at all????????????????

dhalperi, 2015-03-13:
https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection 

you run the receiver code on the receiver nodes, and the transmit code on the transmitters. it could literally not be simpler.

chenyongxi, 2015-03-13:
Sorry，I do not know the principles of the injection mode. I just do what you said . After that I don't know where is transmitters? where I execut the  transmit code? Can you give me some article with the injection mode

dhalperi, 2015-03-13:
In injection mode, every node involved is a Linux machines running the CSI tool. You designate some of those machines as transmitters and some as receivers.

First put all of the receivers in receiver mode, according to the injection instructions. They will be listening for special packets to measure CSI.

Then put all the transmitters in transmitter mode, according to the injection instructions. Then if you send packets from the transmitters they will be heard by all receivers.

This is, as far as I know, the only way to measure multiple links concurrently.

chenyongxi, 2015-03-19:
Excuse me, I have a question to ask you. If your environment has multiple AP, how I collect the ap i want.(multiple APs are 80211n and no password )

dhalperi, 2015-03-19:
associate to that AP and then monitor in normal client mode.

closed 2015-03-24 by dhalperi