dhalperi/linux-80211n-csitool-supplementary #11: Injection script, cannot log transmitted data

ippeb, 2015-03-20:
I have set up two Intel NUCs, each with the specified NIC (Intel WiFiLink 5300 802.11n MIMO). Logging CSI data by pinging a hardware AP from a NUC worked fine. 

In the past few days, I tried to get the injection script working, but one NUC (the "receiver") doesn't log the data that was send by the other NUC ("transmitter"). I have run all instructions from the README in the injection folder, using the values, "64" for the channel and "HT20". 

Analysing the transmitted packets using Wireshark on the receiving side it seems that all criteria for successfully logging packets were fulfilled (from #5 from the FAQs).
- The packet is received correctly.
- The packet is sent using an 802.11n "HT" rate.
- The packet is sent to a hardcoded, fixed address 00:16:ea:12:34:56 from a broadcast node.

Below I have included my ifconfigs and iwconfigs (from both, the receiver and transmitter) and the Wireshark logs of the received samples. 
Any idea why the CSI Tool cannot log the data sent using `sudo ./random_packets 1 100 1`? Any debugging hints?

**the transmitter**
```
>> ifconfig
eth0      Link encap:Ethernet  HWaddr c0:3f:d5:6d:41:e4
          inet addr:10.0.0.238  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::c23f:d5ff:fe6d:41e4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11488 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9658 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5133991 (5.1 MB)  TX bytes:1672958 (1.6 MB)
          Interrupt:20 Memory:f7d00000-f7d20000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:338 errors:0 dropped:0 overruns:0 frame:0
          TX packets:338 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:26037 (26.0 KB)  TX bytes:26037 (26.0 KB)

mon0      Link encap:UNSPEC  HWaddr
00-21-6A-7E-30-A4-3A-30-00-00-00-00-00-00-00-00
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2041 errors:0 dropped:2041 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:366299 (366.2 KB)  TX bytes:0 (0.0 B)
```

```
>> iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

wlan0     IEEE 802.11abgn  ESSID:off/any
          Mode:Managed  Frequency:5.32 GHz  Access Point: Not-Associated
          Tx-Power=15 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

lo        no wireless extensions.
```

**the receiver**
```
>> ifconfig
eth0      Link encap:Ethernet  HWaddr c0:3f:d5:6e:08:88  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:20 Memory:f7d00000-f7d20000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2104 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2104 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:169872 (169.8 KB)  TX bytes:169872 (169.8 KB)

wlan0     Link encap:UNSPEC  HWaddr 00-21-6A-12-8A-D8-3A-30-00-00-00-00-00-00-00-00  
          inet addr:192.168.1.8  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

```
>> iwconfig
eth0      no wireless extensions.

lo        no wireless extensions.

wlan0     IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm   
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
```
Wireshark logs of the received samples (ran `sudo ./random_packets 1 100 1` three times): 
[Wireshark logs](http://people.ee.ethz.ch/~josziegl/wireshark_logs_injection.pcapng)

dhalperi, 2015-03-20:
I'm not certain, but is the receiver really set up correctly? I find it very strange that it has an IP address, for instance. That should not be happening if it is in injection mode.

ippeb, 2015-03-20:
Well spotted, yes, the IP address came from pinging the hardware AP (had modified /etc/network/interfaces ).

I now re-installed Ubuntu and the CSI Tool from scratch on both devices. Here's what I get. 
**transmitter**
```
>> ifconfig
eth0      Link encap:Ethernet  HWaddr c0:3f:d5:6d:41:e4
          inet addr:10.0.0.238  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::c23f:d5ff:fe6d:41e4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:829838 errors:0 dropped:0 overruns:0 frame:0
          TX packets:128445 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1222488604 (1.2 GB)  TX bytes:10548134 (10.5 MB)
          Interrupt:20 Memory:f7d00000-f7d20000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3168 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:334951 (334.9 KB)  TX bytes:334951 (334.9 KB)

mon0      Link encap:UNSPEC  HWaddr
00-21-6A-7E-30-A4-3A-30-00-00-00-00-00-00-00-00
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:697 errors:0 dropped:705 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:132001 (132.0 KB)  TX bytes:0 (0.0 B)
```
```
>> iwconfig
eth0      no wireless extensions.

mon0      IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

wlan0     IEEE 802.11abgn  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

lo        no wireless extensions.
```
**receiver**
```
>> ifconfig
eth0      Link encap:Ethernet  HWaddr c0:3f:d5:6e:08:88  
          inet addr:10.0.0.200  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::c23f:d5ff:fe6e:888/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:841248 errors:0 dropped:0 overruns:0 frame:0
          TX packets:102314 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1249440105 (1.2 GB)  TX bytes:8321296 (8.3 MB)
          Interrupt:20 Memory:f7d00000-f7d20000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1822 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1822 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:183893 (183.8 KB)  TX bytes:183893 (183.8 KB)

wlan0     Link encap:UNSPEC  HWaddr 00-21-6A-12-8A-D8-3A-30-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:65 errors:0 dropped:64 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8624 (8.6 KB)  TX bytes:0 (0.0 B)
```
```
>> iwconfig
eth0      no wireless extensions.

wlan0     IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.
```
Sample of the sent package using Wireshark: (http://people.ee.ethz.ch/~josziegl/wireshark_logs_injection_v2.pcapng)[Wireshark logs v2]

ippeb, 2015-03-20:
It now works! Re-ran everything. Had forgotten to substitute "rmmod" with "modprobe -r". (Had done it before re-installing Ubuntu, forgot it now.)

Thanks so much for your help @dhalperi !

closed 2015-03-20 by ippeb

dhalperi, 2015-03-20:
Rmmod vs modprobe.... Is this another change from Ubuntu 10.04lts? If so, please submit a pull request with a fix!

Glad it's working!!

Boeing738ng, 2015-03-21:
I have same problems as above. Should I reinstall ubuntu and csitools again?

dhalperi, 2015-03-21:
Well, if you're not running the right kernel version, which ended up being the root cause of this problem, you need to update or reinstall the CSI Tool. I don't think you should need to wipe the machine.