dhalperi/linux-80211n-csitool-supplementary #12: Injection, cannot set transmit rate for transmitter

ippeb, 2015-03-20:
After running `./setup_inject.sh 64 HT20` when configuring the transmitter, I am unable to set the transmit rate. There exists no file called "monitor_tx_rate", `find /sys -name monitor_tx_rate` is empty and ``echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate` `` will not work.

```
>> ls /sys/kernel/debug/ieee80211/phy0/iwlwifi/iwldvm/debug/
chain_noise             echo_test   missed_beacon  power_save_status  reply_tx_error  rxon_filter_flags  sensitivity   ucode_bt_stats       ucode_rx_stats  ucode_tx_stats
clear_ucode_statistics  fw_restart  plcp_delta     protection_mode    rf_reset        rxon_flags         txfifo_flush  ucode_general_stats  ucode_tracing
```
Any idea why I am missing this entry? I am also missing the entry "bcast_tx_rate".
Note, I am not associated to any AP, but I don't think I need to be?

network configuration
```
>> ifconfig
eth0      Link encap:Ethernet  HWaddr c0:3f:d5:6d:41:e4  
          inet addr:10.0.0.238  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::c23f:d5ff:fe6d:41e4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:16448 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10458 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:13165558 (13.1 MB)  TX bytes:1692883 (1.6 MB)
          Interrupt:20 Memory:f7d00000-f7d20000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:176 errors:0 dropped:0 overruns:0 frame:0
          TX packets:176 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:13984 (13.9 KB)  TX bytes:13984 (13.9 KB)

mon0      Link encap:UNSPEC  HWaddr 00-21-6A-7E-30-A4-3A-30-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1175 errors:0 dropped:1175 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:112120 (112.1 KB)  TX bytes:0 (0.0 B)

eth0      no wireless extensions.
```

wireless network configuration
```
>>> iwconfig
mon0      IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlan0     IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.
```

dpward, 2015-03-20:
It sounds like the original driver is still getting loaded instead of the modified driver. What is the output of:
```
modinfo iwlwifi | head -n 1
```

This filename should be that of the modified driver in the `/lib/modules/.../updates/` directory. If it is not, check that you ran `sudo depmod` after the installation. Then make sure to unload the original driver:
```
sudo modprobe -r iwlwifi mac80211
```
or alternatively, reboot the system. Please let us know what happens.

dpward, 2015-03-20:
If the above does not resolve the issue, can you also post the output of:
```
grep "CONFIG_IWLWIFI_DEBUG" /lib/modules/$(uname -r)/build/.config
```

dpward, 2015-03-20:
Since `monitor_tx_rate` used to appear on your system, it seems very likely that your kernel was updated. If so, the modified driver has to be re-built. This can be done by repeating step 2 of the installation, but skip the commands that begin with `git clone`, `git checkout`, or `git remote`.

dhalperi, 2015-03-20:
Yes, this looks right. Thanks @dpward.

ippeb, 2015-03-20:
Spot on! I re-installed a fresh Ubuntu image, ran all the installation instructions and I now get 
```
>> modinfo iwlwifi | head -n 1
/lib/modules/3.16.0-31-generic/updates/iwlwifi.ko
```
However, I did run into a little problem when running the setup scripts. I'm mentioning it here, as this may be helpful for other people. Please check, if this also makes somewhat sense. Also, interested if you had encountered similar problems.

**./setup_monitor_csi.sh 64 HT20**
Got the following error `command failed: Device or resource busy (-16)` which was caused by the line `iw wlan0 set channel 64 HT20`. 
Running the following, "worked" for me:  `ip link set wlan0 down`, `ip link set wlan0 type monitor`, `ip link set wlan0 up`, `iw wlan0 set channel 64 HT20`, `ifconfig wlan0 up` .

**./setup_inject.sh 64 HT20**
Got the following error `command failed: Device or resource busy (-16)`, which was caused by the line `iw mon0 set channel 64 HT20`. 
Running the following, "worked" for me: `ifconfig wlan0 down`,  `iw mon0 set channel 64 HT20`, `ifconfig mon0 up` .

Logging data with receiver, transmitter still doesn't work though :(. Please refer to #11 .

closed 2015-03-20 by ippeb

linkin8834, 2015-03-24:
Hello,
     I have the same issue with ippeb  (command failed: Device or resource busy (-16)). if I ran the commands ifconfig wlan0 down, iw mon0 set channel 64 HT20, ifconfig mon0 up in terminal by sequence, it works. However, if I put these three commands into setup_inject.sh, I still have 'Device of resource busy'... Is there any solution to this?

dhalperi, 2015-03-24:
Is there a reason the 4 lines here, which loop until success, don't work for you? https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/injection/setup_monitor_csi.sh#L5

linkin8834, 2015-03-24:
Hello,
       I cannot figure out why this does not work on my PC. I have to use this sequence of commands:
ip link set wlan0 down, ip link set wlan0 type monitor, ip link set wlan0 up, iw wlan0 set channel 64 HT20, ifconfig wlan0 up
to make it work...Anyway I can do it manually. What ippeb posted here is correct in my case.

dhalperi, 2015-03-24:
This may be a backwards-incompatible change when @dpward upgraded from
10.04LTS to 14.04LTS (or whatever it is now) :)

linkin8834, 2015-03-25:
It seems that I solve this problem.
For the setup_inject.sh, you need to turn on the mon0 before setting it. Here it is

ifconfig mon0 up
while [ $? -ne 0 ]
do
    ifconfig mon0 up
done

iw mon0 set channel $1 $2

linkin8834, 2015-03-25:
Think I solve the problem for setup_inject.sh and setup_monitor_csi.sh.
My OS: Ubuntu 14.04
Kernel: 3.13.0-39-generic

'#!/usr/bin/sudo /bin/bash
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi debug=0x40000
if [ "$#" -ne 2 ]; then
    echo "Going to use default settings!"
    chn=64
    bw=HT20
else
    chn=$1
    bw=$2
fi
ifconfig wlan0 2>/dev/null 1>/dev/null
while [ $? -ne 0 ]
do
	        ifconfig wlan0 2>/dev/null 1>/dev/null
done
iw dev wlan0 interface add mon0 type monitor

ifconfig mon0 up
while [ $? -ne 0 ]
do
	       ifconfig mon0 up
done

iw mon0 set channel $chn $bw'

dpward, 2015-03-30:
@dhalperi I've been away on vacation and am just getting caught up. I haven't revised the sample scripts or documentation for injection mode (mainly because those were already somewhat out of date) but I'll go ahead and do that. In particular, we should be favoring the use of the `ip` and `iw` commands, which are designed to replace `ifconfig` and `iwconfig`.

dhalperi, 2015-03-31:
@dpward – I have been traveling as well ;).

Thanks, those recommendations make great sense!