dhalperi/linux-80211n-csitool-supplementary #15: How to force three transmit antennas?

linkin8834, 2015-03-23:
Hello,
         I know that the rate flag would be something like 1[C,D,E,F]xxx for three antenna configuration. I associate my 5300 with my three antenna wireless router. I keep refresh the rate_scale_table and ping the wireless router at another terminal. The rate scale table seems to change all the time (only allow 16 entries to be displayed at the same time). I did not see any rate starts with 1C (meaning that, the wireless router seems to be happy to work with a maximum of TWO transmit antennas). 
        For my current project, I need to use the three transmit antennas at the same time. I tried to put 0x1c113 in the rate scale table, but it crashes the connection (I have to re-associate with the wireless router again everytime I put 0x1c113 into rate_scale_table).
        How can I get a FULL list of supported rate by an AP/station (instead of keep refreshing rate_scale_table and expect one rate starting with 0x1c...)? Thanks!

dhalperi, 2015-03-23:
There are several questions here:

- what is the model of your wireless router? Even if it has 3 antennas, does it support 3 streams? Should be billed as a "450 Mbps" router

- Can you say more about the hardware on the non-router end? In particular, what computer model are you using; did you connect all three antennas; what channel are you using and do all the antennas you connected support the band you're using?

- To get the full set of capabilities for all access points, try `iw list`. Please paste the result here.

linkin8834, 2015-03-24:
Hello,
     1. It was a MERCURY wireless router (I bring it from China), which has three antennas BUT only support 300Mbps (thank you for your remainder!)
     2. It was Ubuntu 14.04, with all three antennas connected. I can see CSIs on all three RXs, so I think the three RX antennas is working as expected.
     3. I am not at the school right now.

     So basically it was due to the wireless router...
     Maybe the only choice is to use injection script... Is the first argument of the bash script the channel number 64 (5.32GHz), and the second one the 802.11n MCS index in http://mcsindex.com/ (HT20, three streams, and 16-QAM 3/4)? 
 
     Thanks so much for your kindly help in the past few weeks!

dhalperi, 2015-03-24:
The two arguments simply configure the channel -- the `20` there is for 20 MHz rather than 40 MHz. You set the rate using `monitor_tx_rate`.

dhalperi, 2015-03-24:
Sounds like we narrowed down the immediate problem with your AP link going down, so I'm going to close the issue

closed 2015-03-24 by dhalperi