dhalperi/linux-80211n-csitool-supplementary #16: sudo iwconfig wlan0 mode monitor 2>/dev/null 1>/dev/null  RETURN 250 ERROR CODE

Boeing738ng, 2015-03-24:
When I set monitor, after I run **sudo iwconfig wlan0 mode monitor 2>/dev/null 1>/dev/null ** in commandline, then I echo $?, it return 250. But I have run command:
1. modprobe -r iwlwifi mac80211 cfg80211
2. modprobe iwlwifi connector_log=0x1
before it.
How to fix it?

dhalperi, 2015-03-24:
Why don't you stop redirecting the output and actually report what it says?

Boeing738ng, 2015-03-24:
Thank you for replying me first!
it outputted as these:

*Error for wireless request "Set Mode" (8B06)*
      *SET failed on device wlan0 ; device or resource busy.*

dhalperi, 2015-03-24:
@dpward this is a problem you've seen before, right? AFAICT this did not happen in 10.04LTS

dhalperi, 2015-03-24:
Is there a reason you can't use the four lines here, which loop until success? https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/injection/setup_monitor_csi.sh#L5

Boeing738ng, 2015-03-25:
If I use while loop here, it will run forever.

closed 2015-03-28 by Boeing738ng