dhalperi/linux-80211n-csitool-supplementary #17: Problem about Injection: I can sent out in mode 0, but not in  mode 1.

Boeing738ng, 2015-03-24:
If I send packages in mode 0, the packages can be received by wireshark correctly. But If I send them out in mode 1, the packages are sent out much quickly than that in mode 0. In addition, the wireshark on the receiver can not receive any packages.
How to fix it?

dhalperi, 2015-03-24:
Much quicker -- is it because the monitor rate is faster?

Cannot be received -- is it because you're using too fast a rate / unsupported rate at the receiver?

Please try debugging these problems before you ask the questions. And if you have debugged the problem, tell us everything you did (at a high level) with all the exact commands you ran start to finish.

Boeing738ng, 2015-03-25:
I am not very familiar with linux networking. Can you give me some instructions about how to debug it?

Boeing738ng, 2015-03-25:
@dhalperi  But if it is right that because the monitor rate is faster, why I can send out the packages in mode 0 normally and the packages sent out can be received by the receiver correctly? The problem about mode 0 is that log_to_file doesn't have any output.

dhalperi, 2015-03-25:
Please re-read FAQ 5 to understand why packets in mode 0 do not measure csi. file:///Users/dhalperi/Code/802.11n/linux-80211n-csitool/faq.html

You did not answer the questions I asked nor did you paste any commands. We simply cannot help without useful information.

Boeing738ng, 2015-03-25:
*the packet is sent to a hardcoded, fixed address 00:16:ea:12:34:56 from a broadcast node. See the injection scripts.*

I can not understand the meaning of *from a broadcast node* here. Could you explain it to me? Thanks!
Here, it just defines the address of the receiver whom we want to send to. I think that the address of the sender have no role in injection. Is it correct?

dhalperi, 2015-03-25:
"address of the sender" is the "from" : sender MAC needs to be the
broadcast MAC, aka ff:ff:ff:ff:ff:ff

Boeing738ng, 2015-03-25:
memcpy(packet->addr1, "\x00\x16\xea\x12\x34\x56", 6);
		memcpy(packet->addr2, "\x00\x16\xea\x12\x34\x56", 6);
		memcpy(packet->addr3, "\xff\xff\xff\xff\xff\xff", 6);
  

According to 802.11n-2009, the address 2 is SA( sender's address) and the address 3 is IBSS. You have said that the sender's address should be ff:...ff, but the addr2 in the mode 1 is set to x00\x16\xea\x12\x34\x56, not ff:....:ff as you said.

dhalperi, 2015-03-25:
Hmm, well okay, maybe the text needs clarification.

Here's [the code](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/injection/random_packets.c#L93).
The `ToDS` and `FromDS` fields are both `0`, so we expect 3 addresses: `DA`, `SA`, `BSSID` in that order. (See `802.11-1999`, Table 4 in section 7.2.2.)

In mode `0`, the `SA` is the card's native MAC address; you can clearly see from FAQ.5, even as confusingly written, that this will not work.

I have updated FAQ.5 slightly to include this clarification.

Boeing738ng, 2015-03-25:
I just installed 2 antennas on the receiver and the transmitter. Does it affect the working of log_to_file? Should I installed the third antenna? Or could I just enable two antenna in monitor_tx_rate code?

dhalperi, 2015-03-25:
Please copy and paste all commands you run at the receiver and all commands you run at the transmitter, plus their outputs.

Please start the log with the command in which you remove the module, probably `sudo modprobe -r iwlwifi`

dhalperi, 2015-03-25:
Also, for clarity, please use [GitHub Flavored Markdown](https://help.github.com/articles/github-flavored-markdown/) to format the logs so they are readable.

Boeing738ng, 2015-03-25:
**On the receiver:**
```
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi connector_log=0x1
ifconfig wlan0 up  #added by me
iwconfig wlan0 mode monitor
iw wlan0 set channel 64 HT20
ifconfig wlan0 up

sudo ../netlink/log_to_file log.dat
```

**On the transmitter**
```
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi debug=0x40000
ifconfig wlan0 
iw dev wlan0 interface add mon0 type monitor
iw mon0 set channel 64 HT20
ifconfig mon0 up

echo 0x1c10e | sudo tee `find /sys -name monitor_tx_rate`
sudo ./random_packets 100000 100 1
```

I have eliminated all sudos before every command.

dhalperi, 2015-03-25:
Okay, so your first trial is with MCS 14 (`0x0e` at the end) and 3 antennas (`0x1c` at the beginning)? That seems wrong...

1. If you only connected two antennas, don't use `0x1c` – use the mask for only the antennas you connected! The hardware shaping the signal in a way that with fewer antennas 1/3 of the power will be lost AND the streams are actually *less* orthogonal to each other than they would be.

2. Why not start with MCS 1 as we suggest in the instructions? MCS 14 is very hard to receive with only two antennas. In fact, we have an unpublished paper to that effect! I'll post it on my website and add a link... If you start with MCS 1 you will be able to distinguish between "the packets are not able to be received because of RF conditions like signal strength" and "there is a software configuration error that makes the CSI measurement not work".

3. Finally, the fact that you're using MCS 14 clearly explains why the packets are sent much faster in mode 1 – you're using a bitrate that's 18x faster than you are in mode 0.

Boeing738ng, 2015-03-25:
Inside
[commands.h](https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-3.13/drivers/net/wireless/iwlwifi/dvm/commands.h#L259)

``` 
High-throughput (HT) rate format for bits 7:0 (bit 8 must be "1"):
 *  2-0:  0)   6 Mbps
 *        1)  12 Mbps
 *        2)  18 Mbps
 *        3)  24 Mbps
 *        4)  36 Mbps
 *        5)  48 Mbps
 *        6)  54 Mbps
 *        7)  60 Mbps
*  4-3:  0)  Single stream (SISO)
 *        1)  Dual stream (MIMO)
 *        2)  Triple stream (MIMO)
```
between bit 3 and 4,  Single stream,Dual stream,or Triple stream, which one should I select?

In addition, I have only two antennas in my receiver? Is there some parameter which should be set to make just two antennas running?

Boeing738ng, 2015-03-25:
How to distinguish a package sent in ht mode or in non-ht mode on wireshark?
[image]
Image above show that the packets sent in mode 0 were received my Macbook.
You can see that there are two channel type. 

In additon, if I can use wireshark to capture injection packages on the receiver?

dhalperi, 2015-03-25:
Commands.h does factor out the rates like that, but it is easiest to just read those 5 bits as the encoding of the MCS. See the 802.11n-2009 standard.

Also, much of this is covered in FAQ.1. Please re-read it! http://dhalperi.github.io/linux-80211n-csitool/faq.html

- MCS 1 is 0x01  (00001 in binary; 00 means single-stream and 001 means 13 Mbps)
- MCS 14 is 0x0e (01110 in binary; 01 means means two-stream and 110 means 58.5 Mbps, for a total of 117 Mbps)

All of these depend on channel width and other physical configurations. I *really* suggest familiarizing yourself with the 802.11n-2009 standard!!

dhalperi, 2015-03-25:
Wireshark – your tool, you need to figure out how to use it. I would look at the field called "MCS information", maybe? MCS is an 802.11n+ specific term.

Boeing738ng, 2015-03-25:
OK, THANK YOU VERY MUCH!
I will try it tomorrow...
oh no, I will try it today. 
it's 02:17 am now.

dhalperi, 2015-03-25:
get some sleep!

dhalperi, 2015-03-25:
Here's the unpublished tech report I mentioned: http://dhalperi.github.io/linux-80211n-csitool/halperin-802.11n-measurement-UW-Tech-Report-2009.pdf

(Now also linked from the [main page of the CSI Tool](http://dhalperi.github.io/linux-80211n-csitool/#publications))

Boeing738ng, 2015-03-26:
The problem stills exists. Here is output of command *ifconfig* and *iwconfig* on the transmitter.

**OUTPUT of ifconfig:**
```
eth0      Link encap:Ethernet  HWaddr 00:1a:80:1a:17:a1  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:16 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:22 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1596 (1.5 KB)  TX bytes:1596 (1.5 KB)

mon0      Link encap:UNSPEC  HWaddr 00-21-6A-10-E3-EA-00-00-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:102018 errors:0 dropped:103464 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20002423 (20.0 MB)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 00:21:6a:10:e3:ea  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2840 errors:0 dropped:10 overruns:0 frame:0
          TX packets:517 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:716020 (716.0 KB)  TX bytes:105265 (105.2 KB)
```

**OUTPUT of iwconfig**
```
wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          
mon0      IEEE 802.11abgn  Mode:Monitor  Frequency:5.32 GHz  Tx-Power=15 dBm   
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          

```

If I doesn't set addr2 to mac address of wlan0, then packages would not be sent out. If ONLY the addr2 is set to the MAC address of wlan0, the packages can be sent out successfully and detected on receiver.

The method I use to judge whether the packages have been sent out is that whether or not the wireless card status light twinkle.
the wireless card status light is the rightmost light on the photo below which is marked with "WLAN"
[image]

Boeing738ng, 2015-03-26:
In addition, OUTPUT of **uname -a** and **lsb_release -a** are shown below
**OUTPUT of uname -a :  **
```
Linux ubuntu 3.5.7-csitool+ #1 SMP PREEMPT Fri Dec 5 14:09:19 CST 2014 i686 GNU/Linux
```
**OUTPUT of lsb_release -a :  **
```
Distributor ID:	Ubuntu
Description:	Ubuntu 10.04.4 LTS
Release:	10.04
Codename:	lucid
```

Boeing738ng, 2015-03-27:
@dhalperi  Thank you!
I have solved the problem.
I reinstalled a 14.04 LTS Ubuntu.
Then, all work very well.

Here is the package I captured from my mac.
[image]

Boeing738ng, 2015-03-28:
close

closed 2015-03-28 by Boeing738ng