dhalperi/linux-80211n-csitool-supplementary #20: Why log_to_file captures no packets

simondchen, 2015-03-30:
Dear Mr. Ward

        I am a undergraduate student in XIDIAN University,china.I am not a native English Speaker,and I am not sure whether I ignore some etiquette or say anything in a wrong way.If so,I apologize for my faults.I am using your csi tool to operate an experiment.I buy two Intel 5300 NICs and then follow the installation instructions on the official website.And I set one as AP and the other as station.While I can ping the AP from the station,I cannot grasb any packets using log_to_file tool.I assume that AP and station are exchanging packets using 802.11a/b/g(legacy) instead of 802.11n(HT).I modify the rate_scale_table by running "echo 0x1c113 | sudo tee /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/00:0f:34:9d:01:a0/rate_scale_table",but it still doesnot work.I really don't know how to fix this problem.
        And the official website point out that injection scripts in monitor mode works better than the AP mode,but it does not give much details about monitor mode,So if it's not too much trouble for you,can you provide more information about how to use injection scripts.

Yours faithfully

gitbub1992, 2015-03-30:
I am a student of XiDIAN too,my email is ivycx1992@gmail.com,maybe we can communicate with each other about CSI

dpward, 2015-05-10:
@simondchen You need to replace `00:0f:34:9d:01:a0` with the actual MAC address of the other Intel 5300 NIC; that value in the FAQ is just an example.

closed 2015-05-17 by dhalperi