dhalperi/linux-80211n-csitool-supplementary #22: Injection can not work continually

Boeing738ng, 2015-03-31:
If the transmitter send injection packages continually in a rate of 100 packets per second, log_to_file on the receive can only run normally for several minutes, then it would not have any output at all. 

I checked usage of the memory, and I found that almost all space of the memory had been buffered. 

I want to get CSI informations for several hours. 
How to fix it?  Is there some parameters I can set to empty buffers automatically?

dhalperi, 2015-04-01:
This likely means that you're injecting data at too high a rate – faster than the NIC is sending the packets. Can you try using the [`delay` parameter](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/injection/random_packets.c#L56) – e.g., send only a single packet per second – and see how long the connection lasts?

Boeing738ng, 2015-04-02:
Thanks, I have solved the problem.

closed 2015-04-02 by Boeing738ng

dhalperi, 2015-04-02:
Thanks! – can you please say more about what the problem ultimately was and how you solved it?

Boeing738ng, 2015-04-03:
@dhalperi 
I just increased scheduling priority of the process *log_to_file*, and then it could run continually without interruption.

koushikam, 2018-02-23:
@Boeing738ng  Could you please explain how did you add the scheduling priority to log_to_file. Even I have the same issue on the buffer space availability. Thanks.