dhalperi/linux-80211n-csitool-supplementary #23: seems angle between two antenna will change with time under 5G environment.

xu-wang11, 2015-04-01:
I use csi tool  to measure csi and collect packages. After that, I get each antenna's csi data, like antenna1, antenna2, antenna3, each is a N(package num) * 30 matrix.
I plot the angle between two antennas. matlab code is like
<code>
plot(angle(antenna1 .* conj(antenna2)))
</code> 
[image]

it seems with time going, the angles difference between two antenna will hava an error of  <b>pi</b>?
Any idea bout this? Thanks.

dhalperi, 2015-04-01:
Did the transmitter antenna configuration change? E.g., because of rate selection?

Because it uses spatial mapping, an Intel transmitter will change the relative angle between antennas at different rates.

dhalperi, 2015-04-01:
Just going to mention issue #6 here – I know you saw it, b/c that's your issue – but for the benefit of those people who find this issue later :)

dhalperi, 2015-04-01:
See https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/matlab/sm_matrices.m

dhalperi, 2015-04-11:
Any update, or can I close this?

xu-wang11, 2015-04-11:
I am trying to figure out. But I think it seems like have no relation with data rate. In my experiment, sometimes, data rate hardly change, sometimes, data rate change with time eithor one value or another.

dhalperi, 2015-04-11:
Can you just make another subplot below that plot of the recorded MCS? You can get it right out of the CSI struct: https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/matlab/read_bfee.c#L21 , which becomes the `rate` field in MATLAB: see FAQ.2B

dhalperi, 2015-04-11:
Also, is the transmitter doing antenna selection? E.g., if the packets are a two-stream MCS 8–15, then the transmitter could be switching from transmit antenna AB to AC or BC as part of the rate adaptation. https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-stable/drivers/net/wireless/iwlwifi/iwl-agn-rs.c#L1547

dhalperi, 2015-04-11:
Note that transmit antenna selection is not explicitly visible at the receiver! You only see it implicitly through predicable, consistent changes in the observed CSI.

xu-wang11, 2015-04-13:
I plot one of my datas' {csi_entry.rate}, it looks like below: 
[image]
the data rate is either 2315 or 268.

xu-wang11, 2015-04-15:
Some more detail about my experiment. I use a router with two antennas as transmiter and NIC5300 as receiver. 
The phase shift between two receive antenna change with time like below:
[image]

The phase shift between two transmiter antennas is ok.
[image]

The Data rate
[image]

From three image above, I think data rate have influence on csi, but is not the reason why phase shift after time.
Since the phase between two transmiter antennas don't change a lot. So, maybe it's not relative to Transmiter Antenna Selection.

Am I right or wrong? Thanks for your help.

dhalperi, 2015-04-15:
How are you getting the CSI? Are you using `get_scaled_csi`, for example?

xu-wang11, 2015-04-16:
Yes. I use get_scaled_csi function.

xu-wang11, 2015-04-16:
<code>
function [ret, cfr_a_array, cfr_p_array] = csi_get_csi(filename,antenna)
usePerm = 0;
%csi_trace = read_bf_file(strcat('csi_nlos_5',num2str(i),'.dat'));
csi_trace = read_bf_file(filename);

cfr_a_array = zeros(length(csi_trace),30); % Amplitude of CFR for all packets
cfr_p_array = zeros(length(csi_trace),30); % Phase of CFR for all packets

cfr_array = zeros(length(csi_trace), 30);

for k = 1:length(csi_trace)
    csi_entry = csi_trace{k}; % for the k_{th} packet
    
  
    csi_entry.csi = csi_entry.csi(1, :, :);
    if usePerm == 1
        csi_perm = csi_entry.perm;
        csi_data = zeros(size(csi_entry.csi));
        for tt = 1:3
            csi_data(1, csi_perm(1, tt), :) = csi_entry.csi(1, tt, :);
        end
        csi_entry.csi = csi_data;
    end
    csi_all = squeeze(get_scaled_csi(csi_entry))'; % estimate channel matrix Hexp-figu
    csi = csi_all(:,antenna); % select CSI for one antenna pair

    cfr = csi;
    cfr_a_array(k,:) = abs(cfr);
    cfr_p_array(k,:) = angle(cfr);
    cfr_array(k,:) = cfr;
end

ret = cfr_array;
</code>

xu-wang11, 2015-04-16:
<pre><code>
function [ret, cfr_a_array, cfr_p_array] = csi_get_csi(filename,antenna)

%csi_trace = read_bf_file(strcat('csi_nlos_5',num2str(i),'.dat'));
csi_trace = read_bf_file(filename);

cfr_a_array = zeros(length(csi_trace),30); % Amplitude of CFR for all packets
cfr_p_array = zeros(length(csi_trace),30); % Phase of CFR for all packets

cfr_array = zeros(length(csi_trace), 30);

for k = 1:length(csi_trace)
    csi_entry = csi_trace{k}; % for the k_{th} packet
    
  
    csi_entry.csi = csi_entry.csi(1, :, :);
    csi_all = squeeze(get_scaled_csi(csi_entry))'; % estimate channel matrix Hexp-figu
    csi = csi_all(:,antenna); % select CSI for one antenna pair

    cfr = csi;
    cfr_a_array(k,:) = abs(cfr);
    cfr_p_array(k,:) = angle(cfr);
    cfr_array(k,:) = cfr;
end

ret = cfr_array;
</code></pre>

dhalperi, 2015-04-16:
This code seems somewhat misguided...

1. Please do not manipulate the CSI entry before `get_scaled_csi` – ALWAYS manipulate it afterwards.
2. When you take the transpose of a complex matrix, you need to use the [`.'` operator](http://www.mathworks.com/help/matlab/ref/transpose.html#buauce2-1)?

dhalperi, 2015-05-10:
Any update here? If not, I will close the issue for inactivity. Thanks

xu-wang11, 2015-05-13:
Maybe it's because I use router as transmiter. I use 5300 NIC as transmiter instead and this problem don't come up. So I will close the issue. Thanks for your help.

closed 2015-05-17 by dhalperi

swashah, 2016-12-04:
@xu-wang11 @dhalperi : Hi, I am trying to collect CSI on two laptops in monitor mode simultaneously while third laptop is injecting packets. I have deployed external antennas with both laptops and distance between two laptops is less than half wavelength so logically this CSI (amplitude) should be very similar (although there could be misalignment between data sets due to packet loss). But when I see the CSI in matlab this similariyt is not there on some subcarriers? Do you have any idea?