dhalperi/linux-80211n-csitool-supplementary #24: Same step but Different outputs?

sunhaosia, 2015-04-01:
Dear , I followed the second issue "How do I process CSI with MATLAB or Octave?".but I got a different result. What's wrong with it ? How to figure it out? THANK YOU!My result is as follow:
The timestamp_low and two ans are different!
>> csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');
>> csi_entry = csi_trace{1}

csi_entry = 

    timestamp_low: 4
       bfee_count: 72
              Nrx: 3
              Ntx: 1
           rssi_a: 33
           rssi_b: 37
           rssi_c: 41
            noise: -127
              agc: 38
             perm: [3 2 1]
             rate: 256
              csi: [1x3x30 double]

>> csi = get_scaled_csi(csi_entry);
>> plot(db(abs(squeeze(csi).')))
>> legend('RX Antenna A','RX Antenna B','RX Antenna C','Location','SouthEast');
>> xlabel('Subcarrier index');
>> ylabel('SNR [dB]');
>> db(get_eff_SNRs(csi),'pow')

ans =

   22.1821   22.2698   22.9007   24.6297
 -156.5356 -156.5356 -156.5356 -156.5356
 -156.5356 -156.5356 -156.5356 -156.5356
 -156.5356 -156.5356 -156.5356 -156.5356
 -156.5356 -156.5356 -156.5356 -156.5356
 -156.5356 -156.5356 -156.5356 -156.5356
 -156.5356 -156.5356 -156.5356 -156.5356

>> csi_entry = csi_trace{20}

csi_entry = 

    timestamp_low: 4
       bfee_count: 91
              Nrx: 3
              Ntx: 3
           rssi_a: 34
           rssi_b: 39
           rssi_c: 39
            noise: -127
              agc: 40
             perm: [2 3 1]
             rate: 272
              csi: [3x3x30 double]

>> csi = get_scaled_csi(csi_entry);
>> db(get_eff_SNRs(csi),'pow')

ans =

       Inf       Inf   32.3435   32.6069
       Inf       Inf   32.4238   32.6822
       Inf       Inf   32.2353   32.5051
   25.4763   25.5262   25.8974   26.8482
   24.6893   24.7490   25.1933   26.5660
   21.9185   22.0303   22.8060   24.6483
    6.5818    8.2321   12.4185   16.2016

dhalperi, 2015-04-01:
I am fairly certain this discrepancy is due to slight changes in the MATLAB code since the FAQ was written. The code has changed to better take into account missing noise measurements and because, originally, the `timestamp_low` field was either not present or not brought into MATLAB.

dhalperi, 2015-04-01:
I will re-run the current version of the code on my laptop a bit later and confirm that I get your answers.

sunhaosia, 2015-04-01:
All right,thank you!I will wait for your result.I will re-run the current version of the code on my laptop a bit later and confirm that I get your answers.

dhalperi, 2015-04-01:
I got the same answers as you, so I think everything is working. I have updated the FAQ to reflect the changes.

Thanks!

closed 2015-04-01 by dhalperi

sunhaosia, 2015-04-02:
Thank you dear Daniel发自网易邮箱大师
            在2015年04月02日 02:28，Daniel Halperin 写道:I got the same answers as you, so I think everything is working. I have updated the FAQ to reflect the changes.

Thanks!