dhalperi/linux-80211n-csitool-supplementary #26: Why The  timestamp_low  changed from 0 to 4? And Can I make it at WIN7_x64?

sunhaosia, 2015-04-02:
You said that:
timestamp_low is the low 32 bits of the NIC's 1 MHz clock. It wraps about every 4300 seconds, or 72 minutes. This field was not yet recorded in the sample trace, so all values will be zero.
BUT why does it change to 4? And what does the "Inf" mean?
Thank you ! 
I also mex the read_bfee.c at my laptop to run the "csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');"But the MATLAB breakdown.My OS is Win7_x64 and the MATLAB 2010b with C compiler Visual Studio 2010 .I have known that someone has succeed to make it at WIN7_x64 .So I just want to know how to make it?
Thank you very much!

sunhaosia, 2015-04-02:
I am the quesion vector of #24 .

dhalperi, 2015-04-02:
The timestamp value in the traces is arbitrary and should be ignored. `0` vs `4` makes no difference ... but I'll update the FAQ again. Thanks!

Hopefully, the timestamp should be valid in *your* measurements :)

closed 2015-04-02 by dhalperi