dhalperi/linux-80211n-csitool-supplementary #27: CSI Measure from Transmitters with Different Wireless Card Manufacturers

Daniluvatar, 2015-04-03:
Dear all

I have a receiver with an **Intel IWL 5300** adapter running over **Ubuntu 10.04**, I was making a test to measure the CSI from packets sent from a wireless card with a different manufacturer than IWL 5300 adapter. I’m using a **Ralink DWA-140** adapter which has the driver **rt2800usb** that supports cfg80211 and mac80211. I’m using the injection scripts for the transmitter and receiver as is pointed out in the link https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection, where just for the transmitter I changed rt2800usb instead of iwlwifi in **setup_inject.sh**.

After running the **random_packets** program, when I check it out in the receiver, I didn’t get the CSI measures for the transmitter with the DWA-140 adapter, but when I test it with the transmitter that has a IWL 5300 adapter I get the CSI measures without any issue.

I wonder if the transmitter needs to be also the Intel IWL 5300 adapter in order to the receiver be able to get the CSI information. Otherwise can you assist me please, I don’t know if I’m missing something, so that the transmission with Ralink DWA-140 adapter works and the receiver is able to get the respective CSI information.

Thanks in advance for your help.

dhalperi, 2015-04-03:
Please review [FAQ.5](http://dhalperi.github.io/linux-80211n-csitool/faq.html) to ensure that you fully understand in what situations the packets are received.

I'm glad you are able to get injection working when both sides are IWL5300! Awesome first step.

So, for the injection packets to be received:

1. Are they sent using an HT rate? (You did not mention above how you set the rate for the rt2800usb packets).
2. Does the rt2800usb actually honor the RA/SA/DA set by the `random_packets` program?
3. Are the packets received successfully by the receiver?

Daniluvatar, 2015-04-06:
Absolutely, I took into consideration the guidelines from [FAQ.5](http://dhalperi.github.io/linux-80211n-csitool/faq.html)

1) Yes, in fact I created a script (script_rt2800usb.sh) based on your [scripts](https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection) for setting the HT as follows:

```sh
#!/usr/bin/sudo /bin/bash

echo Setting mac80211 and cfg80211...
modprobe -r rt2800usb mac80211 cfg80211
modprobe rt2800usb debug=0x40000
echo Disabling wireless interface...
ifconfig wlan0 down
echo Adding monitor interface...
iw dev wlan0 interface add mon0 type monitor
echo Enabling monitor interface... 
ifconfig mon0 up
echo Setting channel $1 and $2...
iw mon0 set channel $1 $2
echo Enabling wireless interface...
ifconfig wlan0 up
echo Done
```
Then, I just set the HT like this: #./script_rt2800usb.sh 1 HT20

2) Yes, the **random_packets** program is executed using *mode 1*. So it should be send with the RA/SA/DA set as it is defined in the program:
```c
(mode == 1) {
		memcpy(packet->addr1, "\x00\x16\xea\x12\x34\x56", 6);
		memcpy(packet->addr2, "\x00\x16\xea\x12\x34\x56", 6);
		memcpy(packet->addr3, "\xff\xff\xff\xff\xff\xff", 6);
	}
```
I even created another script(script_rt2800usb_IntelMAC.sh) to define the MAC of the interfaces to *00:16:ea:12:34:56*. I wanted to be sure that the rt2800usb actually honor the RA/SA/DA.
```sh
#!/usr/bin/sudo /bin/bash

#With this script both wlan0 and mon0 interfaces are set to Intel's MAC

echo Begin MAC modification
echo Disabling wireless interface...
ifconfig wlan0 down
echo Modifying wireless interface MAC...
macchanger --mac 00:16:ea:12:34:56 wlan0
echo Adding monitor interface...
iw dev wlan0 interface add mon0 type monitor
echo Modifying monitor interface MAC...
macchanger --mac 00:16:ea:12:34:56 mon0
echo Enabling monitor interface... 
ifconfig mon0 up
echo Setting channel $1 and $2...
iw mon0 set channel $1 $2
echo Enabling wireless interface...
ifconfig wlan0 up
echo End MAC modification
```
As you can see, the *ifconfig* command shows the interfaces updated to the MAC *00:16:ea:12:34:56*
[image]

And just set the HT like this: #./script_rt2800usb_IntelMAC.sh 1 HT20

In both cases everything is good, I didn't get errors at all, but when I run the **random_packets** and verify the CSI in the receiver, there is no CSI information =(

3) Yes, the packets are received successfully. I can tell you this because I modified the function that generates the payload in the **random_packets** program to be able to send and receive a legible message in the payload, then I open the binary file generated by the **log_to_file** script to verify such message.
So, with this work done I can see the message sent from the IWL5300 and from the DWA-140 adapters. The only difference is that in the IWL5300 the CSI is generated whereas in the DWA-140 the CSI is NOT generated. Any idea why is this happening ?

dhalperi, 2015-04-06:
Thanks for that helpful response.

I'm still not certain how you know that the packets themselves are sent using an HT rate. If you configure the channel for HT20, most devices will still use the 1 Mbps non-HT rate (or 6 Mbps in 5 GHz) for broadcast packets. How are you controlled the **rate** at which the rt2800 sends the packets?

For the iwl5300, we use this line:

```
echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`
```

but that is Intel-specific.

Daniluvatar, 2015-04-09:
What I did, is to use **iw** command for modifying tx HT MCS bitrates, for example:

``
iw dev wlan0 set bitrates mcs-2.4 6
``
``
iw dev mon0 set bitrates mcs-2.4 6
``

I tried with all the MCS indexes which are able to get rates greater than 54 Mbps, but still the same problem, the CSI is not generated.

I also tried with **iwconfig** command, for example:

``
iwconfig wlan0 rate 100M
``
``
iwconfig wlan0 rate 100M
``

But the problem persists, there is no CSI information.

Do you think that it is necessary to modify the driver or the firmware of the DWA-140 adapter to force the HT rate ? I hope there is an easier way to get the HT rate, if you know please let me know.

dhalperi, 2015-04-10:
I doubt that mechanism for setting rates work. the `iwconfig` method *definitely* doesn't.

You will need to ask someone who understand rt2800 how to set the rate for that device. You might try the Linux-wireless mailing list: https://wireless.wiki.kernel.org/en/developers/mailinglists

Daniluvatar, 2015-04-14:
Ok, thank you. I will send an email to LInux-wireless group.

One more question, maybe there is some USB wireless version of the IWL5300 adapter ?

dhalperi, 2015-04-14:
No.

closed 2015-04-17 by dhalperi

swashah, 2016-11-22:
@Daniluvatar : Hi can you please briefly guide me how can I modify random_packets to include a packet seq no. along with packet and then subsequently can read this seq no. along with CSI? Thanks