dhalperi/linux-80211n-csitool-supplementary #28: Dampening of signal at the subcarriers towards the end

avishekfsu, 2015-04-03:
Hello, I have recently started working with the CSI tool, and I have noticed a dampening of the signal towards the edge subcarriers. Here is an example

[image]

In the figure above, the strength of the signal always goes down towards the end subcarriers. I have seen this in tests I ran at other locations as well, but have not been able to find out why this is happening. Am I doing something wrong with my configuration? 

FYI, the figure above is from 1X3 SIMO link with the rate set at 54Mbps.

dhalperi, 2015-04-03:
You are correct; this effect is a result of the hardware's processing and we have to deal with it :)

closed 2015-04-03 by dhalperi

avishekfsu, 2015-04-03:
Thank you very much for your prompt reply. If possible, can you share with me how to handle this edge effect, i.e., reverting the CSI of the edge subcarriers back to their values without the additional change caused by the hardware, because the actual wireless channel should not always be weaker near the edges of the spectrum?

dhalperi, 2015-04-03:
I have never compensated for it.

avishekfsu, 2015-04-04:
OK, I'll have to think of some way to deal with this. Thank you so much for your help !