dhalperi/linux-80211n-csitool-supplementary #29: I failed to download LORCONv1 from http://802.11ninja.net/svn/lorcon/branch/lorcon-old/

sunhaosia, 2015-04-05:
I want to use injection mode，but I failed to download the LORCONv1.Could you help me?Look forward to your favourable reply. I used axel ,and I just got a file named"default".
It is as followed:
<html><head><title>lorcon - Revision 224: /branch/lorcon-old</title></head>
<body>
 <h2>lorcon - Revision 224: /branch/lorcon-old</h2>
 <ul>
  <li><a href="../">..</a></li>
  <li><a href="BUGS">BUGS</a></li>
  <li><a href="COPYING">COPYING</a></li>
  <li><a href="Lindent">Lindent</a></li>
  <li><a href="Makefile.in">Makefile.in</a></li>
  <li><a href="README">README</a></li>
  <li><a href="THANKS">THANKS</a></li>
  <li><a href="TODO">TODO</a></li>
  <li><a href="aclocal.m4">aclocal.m4</a></li>
  <li><a href="airpinject.c">airpinject.c</a></li>
  <li><a href="airpinject.h">airpinject.h</a></li>
  <li><a href="ajinject.c">ajinject.c</a></li>
  <li><a href="ajinject.h">ajinject.h</a></li>
  <li><a href="bcm43xxinject.c">bcm43xxinject.c</a></li>
  <li><a href="bcm43xxinject.h">bcm43xxinject.h</a></li>
  <li><a href="config.guess">config.guess</a></li>
  <li><a href="config.h.in">config.h.in</a></li>
  <li><a href="config.sub">config.sub</a></li>
  <li><a href="config_win32.h">config_win32.h</a></li>
  <li><a href="configure">configure</a></li>
  <li><a href="configure.in">configure.in</a></li>
  <li><a href="cygbuild.sh">cygbuild.sh</a></li>
  <li><a href="docs/">docs/</a></li>
  <li><a href="hapinject.c">hapinject.c</a></li>
  <li><a href="hapinject.h">hapinject.h</a></li>
  <li><a href="ieee80211.h">ieee80211.h</a></li>
  <li><a href="ifcontrol_linux.c">ifcontrol_linux.c</a></li>
  <li><a href="ifcontrol_linux.h">ifcontrol_linux.h</a></li>
  <li><a href="install-sh">install-sh</a></li>
  <li><a href="iwcontrol.c">iwcontrol.c</a></li>
  <li><a href="iwcontrol.h">iwcontrol.h</a></li>
  <li><a href="l2ping80211.c">l2ping80211.c</a></li>
  <li><a href="l2ping80211.h">l2ping80211.h</a></li>
  <li><a href="lorcon.3">lorcon.3</a></li>
  <li><a href="lorcon_decode.c">lorcon_decode.c</a></li>
  <li><a href="lorcon_forge.c">lorcon_forge.c</a></li>
  <li><a href="lorcon_forge.h">lorcon_forge.h</a></li>
  <li><a href="lorcon_packasm.c">lorcon_packasm.c</a></li>
  <li><a href="lorcon_packasm.h">lorcon_packasm.h</a></li>
  <li><a href="ltmain.sh">ltmain.sh</a></li>
  <li><a href="mac80211inject.c">mac80211inject.c</a></li>
  <li><a href="mac80211inject.h">mac80211inject.h</a></li>
  <li><a href="madwifing_control.c">madwifing_control.c</a></li>
  <li><a href="madwifing_control.h">madwifing_control.h</a></li>
  <li><a href="mwnginject.c">mwnginject.c</a></li>
  <li><a href="mwnginject.h">mwnginject.h</a></li>
  <li><a href="mwoldinject.c">mwoldinject.c</a></li>
  <li><a href="mwoldinject.h">mwoldinject.h</a></li>
  <li><a href="nl80211_control.c">nl80211_control.c</a></li>
  <li><a href="nl80211_control.h">nl80211_control.h</a></li>
  <li><a href="orcon.def">orcon.def</a></li>
  <li><a href="p54inject.c">p54inject.c</a></li>
  <li><a href="p54inject.h">p54inject.h</a></li>
  <li><a href="patch/">patch/</a></li>
  <li><a href="rt2500inject.c">rt2500inject.c</a></li>
  <li><a href="rt2500inject.h">rt2500inject.h</a></li>
  <li><a href="rt2570inject.c">rt2570inject.c</a></li>
  <li><a href="rt2570inject.h">rt2570inject.h</a></li>
  <li><a href="rt61inject.c">rt61inject.c</a></li>
  <li><a href="rt61inject.h">rt61inject.h</a></li>
  <li><a href="rt73inject.c">rt73inject.c</a></li>
  <li><a href="rt73inject.h">rt73inject.h</a></li>
  <li><a href="rtlinject.c">rtlinject.c</a></li>
  <li><a href="rtlinject.h">rtlinject.h</a></li>
  <li><a href="stamp-h.in">stamp-h.in</a></li>
  <li><a href="tuntx.c">tuntx.c</a></li>
  <li><a href="tx.c">tx.c</a></li>
  <li><a href="tx80211.c">tx80211.c</a></li>
  <li><a href="tx80211.h">tx80211.h</a></li>
  <li><a href="tx80211_errno.h">tx80211_errno.h</a></li>
  <li><a href="tx80211_packet.h">tx80211_packet.h</a></li>
  <li><a href="wginject.c">wginject.c</a></li>
  <li><a href="wginject.h">wginject.h</a></li>
  <li><a href="wtinject.c">wtinject.c</a></li>
  <li><a href="wtinject.h">wtinject.h</a></li>
  <li><a href="zd1211rwinject.c">zd1211rwinject.c</a></li>
  <li><a href="zd1211rwinject.h">zd1211rwinject.h</a></li>
 </ul>
 <hr noshade><em>Powered by <a href="http://subversion.tigris.org/">Subversion</a> version 1.6.6 (r40053).</em>
</body></html>

sunhaosia, 2015-04-06:
@dpward @dhalperi

sunhaosia, 2015-04-06:
Excuse me，I have the menu /linux80211n-csitool-supplementary/injection and has many files. I want to use the injection mode,but I can't download LORCONv1.However,I can click these files,and they be read  .I want to know if your download means I make the files one by one,copy and paste.Look for your favorite reply.Thank you very much.(I am a Chinese student,I am not good at customs of English .If it is wrong ,please fogive me thank you .better tell me it ^ ^)Also,what's the meaning of LORCONv1?Any other versions?If I don't download the LORVONv1 and DO 
    receiver:
	./setup_monitor_csi.sh 64 HT20
	sudo ../netlink/log_to_file log.dat

    transmitter:
	./setup_inject.sh 64 HT20
	echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`
	sudo ./random_packets 1 100 1
what will happen?

dhalperi, 2015-04-06:
I'm sorry, but this tool is sufficiently complex to use that it requires a minimum level of comfort with open source technologies and programming. We are not able to provide support of the type that you need. I would advise against using this CSI tool until you are more comfortable with Linux, open source, and the command line in general.

closed 2015-04-06 by dhalperi

sunhaosia, 2015-04-06:
How Can I use the injection model?
            在2015年04月07日 00:52，Daniel Halperin 写道:I'm sorry, but this tool is sufficiently complex to use that it requires a minimum level of comfort with open source technologies and programming. We are not able to provide support of the type that you need. I would advise against using this CSI tool until you are more comfortable with Linux, open source, and the command line in general.