dhalperi/linux-80211n-csitool-supplementary #32: Setting rates in client/AP mode

dougf85, 2015-04-10:
Hi

I am trying to setup the csi tool in client/AP mode and so far I have been able to install it properly and log some csi. I have a question about setting data rates/station info. If I want to set the transmit rate to 0x1c113 (as shown in FAQ #1), should I do this at the client or at the AP?   Both client and AP are machines with IWL5300 card in them. Just to be clear, I have hostapd setup on one machine which is my AP, and I use another machine to ping this access point.

Thanks in advance!

dhalperi, 2015-04-10:
Um, you set the transmit rate at the transmitter. What's the confusion?

What did you try, and what happened when you tried it?

If you're measuring CSI at the client, then you want to set the rate at the AP... since the client measures the packets that the AP transmits.

closed 2015-05-17 by dhalperi

xiukey, 2017-01-17:
@dhalperi when I set the rate at the AP, I cannot find the difference between low and high rates.