dhalperi/linux-80211n-csitool-supplementary #33: INJECTION: recv: No buffer space available

Evangeline0, 2015-04-11:
Dear Daniel ,now I got the injection's scripts work correctly, but the receiver receive handreds of packets , then it stops and shows that: recv: No buffer space available .How to figure it out？
Looking for your favourite reply.

dhalperi, 2015-04-11:
You're probably injecting too many packets at too high a rate for the transmitter to keep up. Try adding an inter-packet delay or reducing the number or size of packets.

See also #22

Evangeline0, 2015-04-12:
Thanks for your reply!
I have seen issue#22, it said that he(or she) just increased the scheduling priority of the process log_to_file, but I don't know how to do it. I  want to add a delay,but I don't know where to put?
Can you tell me some details?
Thank you!

dhalperi, 2015-04-12:
google "linux increase scheduling priority" – will probably find you the right thing

injection has a delay parameter – please read the code for the program

Evangeline0, 2015-04-13:
Thanks so much, I have work it out.
I still have a question,  I get the CSI information from multiple 'transmitters' at the same time, how can I separate the packets from different transmitters.
Looking for your favourite reply.
Thank u!

dhalperi, 2015-04-13:
Glad it worked!

Separating out injected CSI from many transmitters is not something the code currently supports; you will need to be clever and perhaps modify the code to do. Maybe put some special bytes in the packet payload to identify each individual transmitter, for example?

closed 2015-04-13 by dhalperi

ozawali, 2015-07-13:
@Evangeline0 Hello , I met some problems when I use the injection mode.  because I cannot download the LORCONv1 or maybe I ignore somethings Thank you very much!

libing64, 2015-10-11:
@dhalperi 
Separating out injected CSI from many transmitters is not something the code currently supports; you will need to be clever and perhaps modify the code to do. Maybe put some special bytes in the packet payload to identify each individual transmitter, for example?

I generate some bytes and send through the transmitter, but I can not find these bytes in the receiver buffer, can you give me some advice about extracting the custom packet?
Here is my code at receiver part,
```
int ret = recv(sock_fd, netlink_buffer, sizeof(netlink_buffer), 0);
	if (ret == -1) {
		perror("netlink recv");
		return ret;
	}
	
	//> test
	printf("\nNLMSG_LEN:  %d", NLMSG_LENGTH(0));
	printf("\nret:  %d\n", ret);
	for(int i = 0; i < ret; i++)
	{
		printf("%02hhx, ", netlink_buffer[i]);
	}

	/* Pull out the message portion and print some stats */
	struct cn_msg *cmsg = (struct cn_msg*)NLMSG_DATA(netlink_buffer);
/* 	printf("received %d bytes: id: %u val: %u seq: %u clen: %d\n", */
/* 			cmsg->len, cmsg->id.idx, cmsg->id.val, */
/* 			cmsg->seq, cmsg->len); */
	*buf = cmsg->data;
	*len = cmsg->len;
```

The csi data is in the netlink_buffer, however, there seems no  my own packet.

dhalperi, 2015-10-11:
You need to record the packet payload (MPDU) in order to see which device transmitted the packet. 

https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/101