dhalperi/linux-80211n-csitool-supplementary #34: Injection succeed！I can set transmitting antenna. How to set SISO？

KOBETRACY, 2015-04-12:
Dear Daniel,I have make the injection script work correctly.And I can set the transmitter number ,got a one-tx & three-rx system .Now I want to know how to set the receiver,and make an SISO system?Looking for your favorite reply.Thank you!

dhalperi, 2015-04-12:
There is a debugfs parameter called `rx_chains_msk` – I believe you should be able to set it before card initialization in order to disable in software some of the RX antennas. (For some reason, the name of this parameter has always been `rx_chains_msk` even though IIRC it's actually an antenna mask. It's been a long time.)

I can't find any scripts that use it, so I'm not sure exactly how it works any more – probably why this parameter is not documented in the FAQ.

You will almost certainly want to set the parameter while the interface is disabled (e.g., before running `ifconfig wlan0 up` or however you enable the interface).

You set it similarly to how you set other debugfs parameters including `monitor_tx_rate`.

KOBETRACY, 2015-04-13:
Thank you !I will try that.And it is really nice when you say IIRC.I think it sounds really cool!I will remember that abbreviation. :)

dhalperi, 2015-04-13:
Please let me know whether it works

KOBETRACY, 2015-04-13:
Hi Daniel，sorry for replying late.I tried it.I found the parameter 'rx_chains_msk' ，but I cannot configure it like monitor_tx_rate .
I set it as followed:
echo rx_chains: 1 antennas, mask 0x7 | sudo tee /sys/kernel/debug/ieee80211/ phy0/iwlwifi/debug/rx_chains_msk

Then,it shows that:
tee: /sys/kernel/debug/ieee80211/phy0/iwlwifi/debug/rx_chains_msk: Invalid argument

I also used "1 antenna" & "2 antennas" to set it, but failed either.

I want to set the parameter as you said,but I can't.Could you tell me some details about that?Thank you!

dhalperi, 2015-04-13:
I believe the argument is a number 0x1–0x7 corresponding to antenna A (0x1), B (0x2), C (0x4), or any combination of them by addition. So for example,

    echo 0x4 | sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/debug/rx_chains_msk

KOBETRACY, 2015-04-13:
Ok I will try it 5 hours later & I will tell you how it works.Thank you! In my opinion, the mask is fixed so I didn't try it yesterday.

KOBETRACY, 2015-04-14:
It worked well.I followed as you said,and figure it out.The SISO system has been done!Thank you very much!By the way,I want to ask you another question, as follow text shows:
    received 93 bytes: id: 254 val: 1 seq: 0 clen: 93
    received 93 bytes: id: 254 val: 1 seq: 0 clen: 93
    ...............................
Does the "id" mean the transmitter "address"? If I want to use multiple transmitters to send packets simultaneously. Can I change the id number to identify the packets from different transmitters?This is my opinion,I don't know whether it will work correctly.Could you tell me something about that?Thank you!

KOBETRACY, 2015-04-14:
Dear Daniel, I also come up with an idea that if only one transmitter and multiple receivers in the system.Different receiver will get all the packets from the transmitter .What do you think this method could solve my problem which I mention in the last responding?Thank you!

dhalperi, 2015-04-14:
That `id` is not something you can use, no. Please look at the code to see how its value is set, rather than asking me – you need to learn how to find these answers for yourself!

For other questions, please start new issues – it sounds like you got receive antenna selection to work, so I'm closing this one!

dhalperi, 2015-04-14:
You will want to read #27, by the way.

closed 2015-04-14 by dhalperi

KOBETRACY, 2015-04-14:
Thank you very much!