dhalperi/linux-80211n-csitool-supplementary #35: Cant understand the rssi figure for a 3X3 link

avishekfsu, 2015-04-14:
Hello,

Firstly, thanks for your help on issue #28 . I had another question regarding the csi collected in a 3X3 link. I have 3 external antennas attached to each of the machines with the intel 5300 card in them, and I set the rate to 0x1c113 as mentioned in your FAQ. When processing the CSI with matlab , this is the output of 
```
db(get_eff_SNRs(csi), 'pow')

       Inf       Inf   34.1344   34.3114
       Inf       Inf   33.9918   34.1745
       Inf       Inf   32.5335   32.7862
   25.6629   25.7107   26.0715   27.2071
   26.3799   26.4205   26.7294   27.7068
   26.6003   26.6389   26.9326   27.7871
   22.5253   22.6320   23.3183   24.6393
```

and the rssi plot for all 3 receiving antennas for 1 transmitting antenna looks like this.

[image]

Is it normal to see the rssi values fluctuating like this? It almost looks like a periodic curve. Just to be double check, I also plotted out the absolute values of these complex numbers, and the plot looks very similar to this. Also attached the output of csi_entry for a packet in csi_trace

```
csi_entry = 

    timestamp_low: 283160501
       bfee_count: 2000
              Nrx: 3
              Ntx: 3
           rssi_a: 33
           rssi_b: 36
           rssi_c: 38
            noise: -94
              agc: 51
             perm: [3 2 1]
             rate: 275
              csi: [3x3x30 double]
```
I ran this test in an indoor environment (in a typical office room setting) and the machines were fairly close to each other (~2-3meters). I have not seen this kind of behavior in SIMO links, but have not been able to pinpoint if I'm doing anything wrong. Have you ever come across this kind of trend? 

P.S. I also slowed down the rate using 0x1c110( MCS 16) to see if it was a rate issue but I am getting similar figures when I plot out the rssi.

Thank you.

dhalperi, 2015-04-14:
Yes, this is totally normal with the combined effects of spatial mapping and frequency-selective fading.

Make sure you understand what spatial mapping is! You can compare packets using rates:

* 0x1c101
* 0xc101
* 0x4101

if you want to see how Intel's spatial mapping works.

Note that you can use the `get_scaled_csi_sm` function, assuming an Intel 5300 transmitter, if you want the CSI between *TX-RX antennas* as compared with *TX streams using the same antenna configurations-RX antennas*.

avishekfsu, 2015-04-15:
Thank you so much ! I had an older version of the matlab directory and I never came across this function. I pulled the current version and it works perfectly. Thanks again !

closed 2015-04-15 by avishekfsu