dhalperi/linux-80211n-csitool-supplementary #36: MATLAB：The same packets, but different csi

Evangeline0, 2015-04-16:
Dear Daniel：Thanks for your  answering my last question. Now , I gained CSI in MATLAB . But I realized CSI values which are from the same packet,same subcarrier are different ,just like this：
csi：                                                          csi_entry.csi：
val(:,:,1) = 2.7310 - 6.8276i                      val(:,:,1) =4.0000 -10.0000i
val(:,:,2) = -4.7793 - 6.1449i                     val(:,:,2) =-7.0000 - 9.0000i
......                                                            ......
I have read the code roughly, but I don't know why they are different?
It is kind of you if you can tell me something about that. Thank you !

dhalperi, 2015-04-16:
I don't understand what you pasted.

KOBETRACY, 2015-04-17:
Dear Daniel ,I think she or he want to know why csi_entry.csi and csi has different value in the same packet .I check them and also realize this condition.Why they are different? Always，the value of csi_entry.csi is integer,I found.

dhalperi, 2015-04-17:
Please read the FAQ:

> Now that we've described all the fields of this struct, we need to put them all together to compute the CSI in absolute units, rather than Intel's internal reference level. In particular, we need to combine the RSSI and AGC values together to get RSS in dBm, and include noise to get SNR. If there is no noise, as in the sample case, we instead use a hard-coded noise floor of -92 dBm. We use the script `get_scaled_csi.m` to do this:

The ratio of the real/imaginary components is the same:

```python
>>> 2.7310/-6.8276
-0.3999941414259769
>>> 4.0/-10
-0.4
```

the values have simply been scaled to be on a meaningful reference.

dhalperi, 2015-04-17:
Thanks for clarifying, @KOBETRACY !

closed 2015-04-17 by dhalperi