dhalperi/linux-80211n-csitool-supplementary #40: Different place ,the value of CSI similarity

Evangeline0, 2015-04-28:
Dear Daniel ! When finishing the experiment with 1 injection and 2 monitors, I can obtain the data of CSI1 and CSI2 successfully. But a problem perplexes me.
In the experiment, the monitor1 is 1.8m from the injection and the monitor2 is 2.8m. I can obtain the CSI1 and CSI2 from the monitors respectively, and the whole experiment is operated under same conditions. However, the difference between the CSI1 and CSI2 ranging from 1 to15,after calculating,I found when the value of difference is less than 10,the CSI we obtain will be incorrect. I wander what reason leads to this, the hardware’s or software’s? Could you do me a favor to confirm it? 
Thanks very much.

dhalperi, 2015-05-10:
Sorry for the delayed response – I have been traveling for the past month :)

- Can you please say more about the calculations you are doing? What are you computing, how are you computing it, and what are the units on numbers like 1 to 15?
- What do you mean that the CSI will be incorrect; how do you know what it should be?

Evangeline0, 2015-05-11:
Briefly, I want to use the value of CSI to confirm the distance between transmitter and receiver by free space path loss propagation model. In this model, there is a parameter----n, which is the path loss fading exponent and it should be not less than 2. But in my experiments, I use the value of CSI and the distance to compute the n, I found it is about 1, this must be the mistake.
For example:
The reference: d1=3.5m   CSIeff1=35;  d2=4.5m  CSIeff2=24; n=2.2;
my experiment: d1=2.5m CSIeff1=22;  d2=3.6m  CSIeff2=19; n=0.96;
The model comes from the paper----FILA: Fine-grained Indoor Localization
d=[1/(4π)]*[(c/(f0*CSIeff))^2*σ]^(1/n)   d is the distance between tx and rx; c,f0,σ are constant; n is the path loss fading exponent.
Thank you !
@dhalperi

chenyongxi, 2015-05-12:
I'm very glad to see the problem in this, I also in the realization of this formula .But I found my eff value fluctuation is larger .May I know your processing eff worth code? Or the eff values about the range?

dhalperi, 2015-05-12:
These types of measurements are very sensitive to the experimental conditions. I would not make any substantial conclusions without data from, say, twenty points.

And if you have a question about reproducing the analysis from someone else's paper, please ask them,! We are not the experts on their work :)

closed 2015-05-12 by dhalperi

gitbub1992, 2015-05-15:
@Evangeline0 @chenyongxi  I am doing some research on this paper too,maybe we can connect with each other via email.

lidongmeng, 2015-06-08:
@gitbub1992 @chenyongxi @Evangeline0  i am doing same research too, can we communicate with emails??  when I ping the ap, some times I got the same number of packet of the ping number, but sometimes I got more packets than the number of ping times? Is it normal? lidongmeng0213@163.com