dhalperi/linux-80211n-csitool-supplementary #41: installation fine csi.dat is created, now is there any step by step procedure available to do data processing

ersammah, 2015-05-10:
i can't read it the csi.dat, which file from supplementary data should be used to read csi information from it. Is there any readme like well defined installation instructions.

dpward, 2015-05-10:
Yes, this is documented in [FAQ 2](http://dhalperi.github.io/linux-80211n-csitool/faq.html). Replace `sample_data/log.all_csi.6.7.6` with the path to your `csi.dat` file.

dhalperi, 2015-05-10:
Thanks @dpward . @ersammah, please comment further on this issue if you have trouble reading the CSI file with MATLAB and the scripts.

ersammah, 2015-05-10:
I am getting only two RSSI values and the third is always 0, I checked the NIC connections it has 3 antennas connected, can we check whether all three antenna are enabled or not, or is there anything I am missing.

the csi parameter in the structure is also 1x2x30 whereas if it is getting the third value also then it should be 1x3x30. one antenna inside laptop is properly place and other two are very close to each other and NIC is this could be an issue for not getting third rssi.

dhalperi, 2015-05-10:
What Wi-Fi channel are you using? Does the antenna support that channel?

A common problem I found was that several laptops had three antennas, but the third one was a 'WiMAX' antenna and only worked on the 2.4GHz band – so it would be automatically disabled by the hardware and would disappear if using channels in the 5 GHz spectrum.

ersammah, 2015-05-10:
current freq 2.4 GHz and channel is 13. How to know if the antenna supporting the channel?
I am using 2.4 ghz and two rssi values are available and third is not.

dhalperi, 2015-05-10:
What is the value of the `rate` field in MATLAB?

dhalperi, 2015-05-10:
You might wish to read the code that detects "disconnected antennas" and disables them. Maybe add some debug prints to see if it is getting triggered.

https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-stable/drivers/net/wireless/iwlwifi/iwl-agn-calib.c#L773

ersammah, 2015-05-10:
8455

dhalperi, 2015-05-12:
How did your investigation of the disconnected antennas code  and behavior go?

ersammah, 2015-05-12:
Still not able to identify, did not get how to get working antenna from the code you mentioned.

dhalperi, 2015-05-12:
Did you add debug prints and recompile etc? You will need to be proactive
in debugging your setup.

ersammah, 2015-05-13:
After adding debug prints which part do i need to recompile?

dhalperi, 2015-05-14:
Use the instructions in the kernel to reinstall and reboot. Alternately you
can read online about how to dynamically reload modules.

ersammah, 2015-05-14:
observing all the packets csi stored, in some of the packets all three antenna are transmitting, it means they all are connected, but third csi is available in very few packets, is the reason known to u?
I keep on running youtube video while doing so.

ersammah, 2015-05-14:
I am getting the data from one antenna, no problem I will manage for now, but this is for subcarriers of single stream, also is there any way to get multipath information out of this tool? I am much interested in observing the multipath effect on the movement of receiver and TX. Please suggest.

dhalperi, 2015-05-17:
It is very unusual for the card to report CSI but not use all connected antennas, though it is possible: One antenna may have such low SNR that the software decides not to use it.

Usually this happens either because of:

1. physical problems – e.g., the antenna is not connected properly to the cable or to the NIC
2. antenna is not matched to frequency
3. antenna is poorly places in the laptop
4. unlikely, but very bad spatial fading affecting only one antenna

The best way to debug these issues is via physical inspection and re-connection of the devices, plus adding debug code and messages to the kernel driver, recompiling and looking at the logs. You seem unwilling to actually properly debug your setup, so there's not a lot we will be able to help with...

closed 2015-05-17 by dhalperi