dhalperi/linux-80211n-csitool-supplementary #44: No CSI logging

TadiGuy, 2015-05-21:
My installation went well with exception of merging "the driver modifications into the source tree for your specific kernel version" I couldn't merge but didn't pursue further because it seems like an optional step (let me know if it's absolutely required).

I'm able to connect to an 802.11n AP and verified that IWL5300 is operating in HT mode, however I don't get any CSI logs on the console or in the csi.dat file. I see this message in syslog __`iwlwifi: unknown parameter 'connector_log' ignored.`__ so it seems to be ignoring my log filter. My full log is below. In /lib/firmware the original iwlwifi firmware is appropriately renamed with .orig and there is symlink to sigcom2010 version of the firmware so I suppose it is loading the correct firmware. It seems like I still missed something in the build, but not sure what it is. I'll appreciate any help.

<h4> syslog after "sudo modprobe iwlwifi connector_log=0x1" </h4>
kernel: [ 5029.357668] cfg80211: Calling CRDA to update world regulatory domain
kernel: [ 5029.370641] cfg80211: World regulatory domain updated:
kernel: [ 5029.370652] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
kernel: [ 5029.370659] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
kernel: [ 5029.370664] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
kernel: [ 5029.370669] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm)
kernel: [ 5029.370674] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
kernel: [ 5029.370679] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
**kernel: [ 5029.374911] iwlwifi: unknown parameter 'connector_log' ignored**
kernel: [ 5029.375982] Intel(R) Wireless WiFi driver for Linux, in-tree:
kernel: [ 5029.375992] Copyright(c) 2003-2013 Intel Corporation
kernel: [ 5029.376406] iwlwifi 0000:04:00.0: can't disable ASPM; OS doesn't have ASPM control
kernel: [ 5029.376522] iwlwifi 0000:04:00.0: irq 42 for MSI/MSI-X
kernel: [ 5029.376936] iwlwifi 0000:04:00.0: Direct firmware load failed with error -2
kernel: [ 5029.376944] iwlwifi 0000:04:00.0: Falling back to user helper
kernel: [ 5029.380796] iwlwifi 0000:04:00.0: request for firmware file 'iwlwifi-5000-5.ucode' failed.
kernel: [ 5029.380874] iwlwifi 0000:04:00.0: Direct firmware load failed with error -2
kernel: [ 5029.380883] iwlwifi 0000:04:00.0: Falling back to user helper
kernel: [ 5029.384739] iwlwifi 0000:04:00.0: request for firmware file 'iwlwifi-5000-4.ucode' failed.
kernel: [ 5029.384817] iwlwifi 0000:04:00.0: Direct firmware load failed with error -2
kernel: [ 5029.384826] iwlwifi 0000:04:00.0: Falling back to user helper
kernel: [ 5029.387906] iwlwifi 0000:04:00.0: request for firmware file 'iwlwifi-5000-3.ucode' failed.
kernel: [ 5029.388859] iwlwifi 0000:04:00.0: Firmware has old API version, expected v5, got v2.
kernel: [ 5029.388872] iwlwifi 0000:04:00.0: New firmware can be obtained from http://www.intellinuxwireless.org/.
kernel: [ 5029.389314] iwlwifi 0000:04:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm
kernel: [ 5029.452419] iwlwifi 0000:04:00.0: CONFIG_IWLWIFI_DEBUG disabled
kernel: [ 5029.452433] iwlwifi 0000:04:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
kernel: [ 5029.452441] iwlwifi 0000:04:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
kernel: [ 5029.452451] iwlwifi 0000:04:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
kernel: [ 5029.452555] iwlwifi 0000:04:00.0: L1 Disabled; Enabling L0S
kernel: [ 5029.483843] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
kernel: [ 5029.546324] iwlwifi 0000:04:00.0: L1 Disabled; Enabling L0S
kernel: [ 5029.549378] iwlwifi 0000:04:00.0: Radio type=0x0-0x2-0x0
kernel: [ 5029.688763] iwlwifi 0000:04:00.0: L1 Disabled; Enabling L0S
kernel: [ 5029.691766] iwlwifi 0000:04:00.0: Radio type=0x0-0x2-0x0
kernel: [ 5029.720625] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
avahi-daemon[568]: Joining mDNS multicast group on interface wlan0.IPv4 with address 10.0.0.1.
avahi-daemon[568]: New relevant interface wlan0.IPv4 for mDNS.
avahi-daemon[568]: Registering new address record for 10.0.0.1 on wlan0.IPv4.

dpward, 2015-05-21:
The message you highlighted indicates that the modified driver (kernel module) isn't getting loaded; the original one is. Can you run the commands again in the last block of step 2 in the installation instructions?

TadiGuy, 2015-05-21:
Thanks a lot for the quick reply. That was the problem. I reviewed my original installation transcript carefully and noticed that **`sudo depmod cd ..`** part of the commands were truncated after I entered passwd for sudo. Re-running the whole last step solved the problem. Thanks a million for the great tool and support.

BTW:
I'm just curious to know why the logs I posted showed **` loaded firmware version 8.24.2.12 op_mode iwldvm`** which is the customer firmware, but it still didn't work. When I load the original one I get a **` ... version 8.83.5.1..`**.

closed 2015-05-21 by TadiGuy

dpward, 2015-05-21:
@TadiGuy The modified firmware needs some interface to provide the CSI to the user, and you need the modified driver to connect to that interface.