dhalperi/linux-80211n-csitool-supplementary #45: Reloading with CSI logging enabled

hmlsi, 2015-05-25:
To the best of my knowledge, I have taken care of all instructions well. I am facing an issue at step 5. After unloading driver, wi-fi is disconnected and when I reload it with CSI logging enabled (sudo modprobe iwlwifi connector_log=0x1), wi-fi is not being connected. I am stuck at this point, don't know what is doing wrong. Please help

dpward, 2015-05-25:
Are you trying to reconnect to an encrypted AP? See [FAQ #7](http://dhalperi.github.io/linux-80211n-csitool/faq.html).

hmlsi, 2015-05-27:
Miilion thanks for prompt response.Yes, I was doing so. While waiting for your answer, I tried some optional commands.
sudo apt-get install iw
echo iface wlan0 inet manual | sudo tee -a /etc/network/interfaces
sudo restart network-manager
and
echo blacklist iwldvm | sudo tee -a /etc/modprobe.d/csitool.conf
echo blacklist iwlwifi | sudo tee -a /etc/modprobe.d/csitool.conf
now, wi-fi networks are not being showed since I have blacklisted driver ( probably). How to overcome this issue?

hmlsi, 2015-05-27:
Wi-Fi is enabled (enable wi-fi option is checked) but available wi-fi networks are not shown.

hmlsi, 2015-05-27:
when I run 
"sudo iw dev wlan0 scan"
I get 
"command failed: Network is down (-100)"

dpward, 2015-05-27:
There are several things going on:

* You disabled NetworkManager from controlling the wireless interface. That means you can't use the NetworkManager GUI to connect to a wireless network.
* If you blacklist the kernel module, you have to explicitly load the module whenever you reboot the system, using the command from step 5 (`sudo modprobe iwlwifi connector_log=0x1`). This is not your issue though, because the wireless interface *wlan0* still exists on your system. If it did not, the error would have been `No such device (-19)`.
* The error you are seeing is `Network is down (-100)`. This appears because the *wlan0* interface has not been brought up, which you have to do manually if you are using the command line utilities to set up the wireless interface. To do this: `sudo ip link set wlan0 up`

hmlsi, 2015-05-28:
That helped me to connect to wi-fi but I was unable to ping. I have one confusion. In step 2, where we are cloning :
git clone git://github.com/dhalperi/linux-80211n-csitool.git
If I have already downloaded zip file from github and try to run
git checkout ${CSITOOL_KERNEL_TAG}
after entering linux-80211n-csitool directory, It gives me error. "fatal: Not a git repository (or any of the parent directories): .git" Why is it so?
Thanks for your patience. I am new thats why I am having much difficulty in understanding.

dpward, 2015-05-28:
The zip file download from Github does not contain the git repository data. Please follow the instructions and use `git clone` to obtain the source code.

hmlsi, 2015-05-29:
Hi,
Now everything is working fine. I can see available wi-fi networks using 
"sudo iw dev wlan0 scan"
but the problem is I cannot connect to any open network. When I try to connect, wi-fi sign keeps on flashing but ultimately message appears "Disconnected. you are now offline." what am I doing wrong? This is happening after unloading the driver and reloading with CSI enabled. (this time, network is not encrypted)

hmlsi, 2015-05-29:
Now, the situation is, when i run command
"sudo iw wlan0 connect -w hmlsi"
and then
"iw dev wlan0 link"
it shows me
"Connected to 14:cc:20:f6:ac:82 (on wlan0)
	SSID: hmlsi
	freq: 2437
	RX: 200423 bytes (1310 packets)
	TX: 166 bytes (4 packets)
	signal: -60 dBm
	tx bitrate: 1.0 MBit/s

	bss flags:	short-preamble short-slot-time
	dtim period:	1
	beacon int:	100"
but its just temporary, not a long lasting connection. Meanwhile, Network manager GUI shows network is disconnected. my network "hmlsi" is working perfect on windows. 
What is happening?

hmlsi, 2015-05-29:
Plus, at this point, if I want to load my previous drivers with CSI logging disabled, what should I have to do?
Thanks a lot for your patience and timely response

dpward, 2015-05-29:
I thought you had disabled NetworkManager from controlling the wireless interface? You can't control the wireless interface both from NetworkManager and from the command line &mdash; you have to pick which one you are going to use. If you're going to use the command line tools, then you must disable NetworkManager from controlling the interface, otherwise you'll have exactly the behavior you are seeing. If NetworkManager is disabled then the NetworkManager GUI won't show you when you are connected &mdash; you have to rely on the command line tools to tell you.

dpward, 2015-05-29:
You can temporarily switch back to the original driver and firmware as follows:

dpward, 2015-05-29:
You can temporarily switch back to the original driver and firmware as follows.

Restore the original firmware:
```
sudo rm /lib/firmware/iwlwifi-5000-2.ucode
for file in /lib/firmware/iwlwifi-5000-*.ucode.orig; do sudo mv ${file} ${file%.orig}; done
```

Restore the original driver:
```
sudo rm -f /lib/modules/$(uname -r)/updates/{iwlwifi,iwldvm,iwlmvm}.ko
sudo depmod
```

Then later you can do the following:

dpward, 2015-05-29:
You can temporarily switch back to the original driver and firmware as follows.

Restore the original firmware:
```
sudo rm /lib/firmware/iwlwifi-5000-2.ucode
for file in /lib/firmware/iwlwifi-5000-*.ucode.orig; do sudo mv ${file} ${file%.orig}; done
```

Restore the original driver:
```
sudo rm -f /lib/modules/$(uname -r)/updates/{iwlwifi,iwldvm,iwlmvm}.ko
sudo depmod
```

Then later, you can reinstall the driver by just repeating these steps from the installation instructions:
```
cd linux-80211n-csitool
sudo make -C /lib/modules/$(uname -r)/build M=$(pwd)/drivers/net/wireless/iwlwifi INSTALL_MOD_DIR=updates \
    modules_install
sudo depmod
cd ..
for file in /lib/firmware/iwlwifi-5000-*.ucode; do sudo mv $file $file.orig; done
sudo ln -s iwlwifi-5000-2.ucode.sigcomm2010 /lib/firmware/iwlwifi-5000-2.ucode
```

hmlsi, 2015-05-29:
So, you are saying, if I am going to control through network manager then there is no need to unload and reload with CSI enabled (in step 5). I just have to begin logging to CSI file after step 4. right?

dpward, 2015-05-29:
No, that's not correct. If you want to collect CSI, you have to load the driver with `connector_log=0x1` regardless.

The difference is after the driver is loaded. You can either connect to a network using NetworkManager, or disable NetworkManager and use the command-line utilities to connect to a network. (Or if you are using monitor mode you don't need to connect to a network.)

dpward, 2015-05-29:
No, that's not correct. If you want to collect CSI, you have to load the driver with `connector_log=0x1` regardless.

There are different ways to use the adapter after the driver is loaded. Note, this is not specific to CSI Tool &mdash; the same applies with the original driver/firmware. You can either:
 * connect to a network using NetworkManager, or
 * disable NetworkManager and use the command-line utilities to connect to a network, or
 * disable NetworkManager, set the adapter to operate in a different mode (AP mode, monitor mode, etc.), and use it accordingly

hmlsi, 2015-05-29:
I did not disable network manager. that was wrong?
so, I have to disable it using

sudo apt-get install iw
echo iface wlan0 inet manual | sudo tee -a /etc/network/interfaces
sudo restart network-manager
 
and then connect to a network using 

sudo iw wlan0 connect -w <device name>

dpward, 2015-05-29:
See my last post. You have different options for how you can use the adapter. We recommend using the command line, but it's up to you.

dhalperi, 2015-05-29:
Thanks @dpward!

For my own 2 cents, I *strongly* recommend disabling/uninstalling NetworkManager and using the command line utilities. I just don't trust the auto-reconfiguring software not to mess with all the settings and options you set when running an experiment.

hmlsi, 2015-06-05:
Thank you so much for your kind response @dpward  and @dhalperi. 
I have disabled network manager and using command line utilities. Here is what I am getting by running " sudo iw wlan0 connect -w hmlsi"

wlan0 (phy #0): connected to 14:cc:20:f6:ac:82

and when I run "iw dev wlan0 link", I got

Connected to 14:cc:20:f6:ac:82 (on wlan0)
	SSID: hmlsi
	freq: 2437
	RX: 139010 bytes (905 packets)
	TX: 4879 bytes (27 packets)
	signal: -48 dBm
	tx bitrate: 2.0 MBit/s

	bss flags:	short-preamble short-slot-time
	dtim period:	1
	beacon int:	100
Problem: If I look on router side, router statistics show that IP address 0.0.0.0 is being assigned to my device. If I connect any other device, router is assigning IP address correctly. What can be the possible reason? when I try to ping my router, I get the error "network is unreachable". The same router is behaving normally for other devices.
Sorry If I am asking foolish questions. I am new and I am tangled in these issues. Much thanks for your kind and prompt responses

hmlsi, 2015-06-05:
Update : now that problem is solved by editing /etc/network/interfaces. I can ping to my AP now. In instructions, there is written that
"After a few dozen pings, the rate selection should automatically switch to HT (802.11n) bitrates. At this point, log_to_file will start recording packets and the iwlwifi driver will print messages in the kernel syslog."
what type of message will be?dozen of pings mean I have to ping few dozens of time explicitly or just receive and transmit dozens of packet? When I run "sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat" and ping my AP at different terminal, I have to terminate logging of CSI to a file. Is it normal to terminate it (Ctrl+C) or it should be closed automatically?

dpward, 2015-06-05:
Both the AP and the Intel card (independently) perform rate adaptation to adjust the rate that is used when transmitting to the other. Generally the first few packets are transmitted at the lowest possible rate (1 Mbps, from the original 802.11 standard). If the rate adaptation algorithm determines that the channel can support a higher rate, then it should begin to increase it (and vice versa).

The CSI Tool can only collect channel information when it receives packets at a High Throughput (802.11n) rate. When exactly this will happen depends both on the specific rate adaptation algorithm in the AP, as well as your channel conditions.

In client mode, you only communicate directly with the AP: you do not transmit directly to other clients connected to the same AP. Therefore any packets that you send over the wireless interface, regardless of their destination or the type of packet, should influence the AP's rate adaptation.

If your AP has an "802.11n-only" mode, then selecting that would cause it to only transmit at HT rates; then you wouldn't have to wait for the AP's rate adaptation algorithm to select an HT rate, and the Intel adapter would start collecting CSI immediately.

Note that the modified driver for the Intel adapter allows you to override the rate that it uses to transmit (see FAQ 1). As far as collecting CSI, this is only useful if you are communicating directly between two or more Intel adapters. This would mean that you have one of the Intel adapters operating in AP mode, or you are using a different mode such as monitor mode or IBSS mode to form a wireless network.

dpward, 2015-06-05:
You do need to terminate `log_to_file` when you are done collecting CSI (pressing Ctrl+C will do this).

hmlsi, 2015-06-10:
Hi, I am using Octave to process CSI. When I tried to read sample file I got the error.

csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');

error: bash: syntax error near unexpected token `('

It may sound silly but I don't know about how ubuntu works or what to do. Any help is highly appreciated in this regard.
Thanks

dpward, 2015-06-10:
It appears that you didn't run the command in Octave; you ran it from a Bash shell. You need to start Octave before running this command.

hmlsi, 2015-06-10:
now the error is (by running command in octave):
error: 'read_bfee' undefined near line 66 column 21
error: called from
    read_bf_file at line 66 column 19
I apologize if its so obvious to overcome this issue.

dpward, 2015-06-10:
You will need to rebuild the MEX-file if you are using Octave. (This will require the `liboctave-dev` package to be installed in Ubuntu.) To do this, run:
```
mkoctfile --mex read_bfee
```

Then try again.

Also please see #48 &mdash; Octave does not currently implement the `db` function, which is from the MATLAB Signal Processing toolbox, and some of the scripts currently call this function.

hmlsi, 2015-06-10:
output of suggested command:

mkoctfile: unrecognized argument read_bfee
warning: mkoctfile exited with failure status
warning: called from
    mkoctfile at line 171 column 5

I am using octave 4.0.0. I tried to add liboctave-dev package using "sudo apt-get install liboctave-dev" but this also ended with error. I think this is due to conflict of octave version and this package as error message contained following lines too:

liboctave-dev depends on octave (= 3.8.2-4); however:
  Package octave is not configured yet.

dpkg: error processing package liboctave-dev (--configure):
 dependency problems - leaving unconfigured

dpward, 2015-06-10:
Thanks for the update!

hmlsi, 2015-06-11:
1. I am connecting to an AP
2. I have set my AP to operate in 11n mode only
3. Pings are being received successfully on both ends
4. When I begin logging CSI to a file and meanwhile I ping on other terminal, some messages are being printed in the first terminal

received 93 bytes: id: 26 val: 1 seq: 0 clen: 93
wrote 93 bytes [msgcnt=0]
received 93 bytes: id: 26 val: 1 seq: 0 clen: 93
received 93 bytes: id: 26 val: 1 seq: 0 clen: 93
received 93 bytes: id: 26 val: 1 seq: 0 clen: 93
received 93 bytes: id: 26 val: 1 seq: 0 clen: 93

After this, I am trying to read my csi data as mentioned in FAQ#2. (Sample data worked fine)

csi_trace = read_bf_file('log_to_file')

the output is:

csi_trace = {}(0x1)
 
This means null matrix. Why it is so?
(I have copied  read_bf_file.m and read_bfee.c in netlink folder. I thought these are necessary to read csi file or perhaps I am doing this thing wrong.)

dpward, 2015-06-11:
The parameter to read_bf_file() is the file that you *saved* with `log_to_file`. Do not pass 'log_to_file' itself; that is the executable.

You don't need to copy additional files into the netlink folder.

dpward, 2015-06-11:
If you call:
```
linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat
```

Then in MATLAB/Octave use:
```
csi_trace = read_bf_file('csi.dat')
```

dhalperi, 2015-06-11:
That's right! To be clear:

1. Start `log_to_file` with a file parameter like `csi.dat`, just as @dpward suggested.
3. After pinging, `log_to_file` should start printing messages, like the ones you saw. (93 bytes is a 1-stream CSI, if I recall correctly.) This is a good sign!
4. Hit `Control-C` to stop `log_to_file`, which should flush files to disk.
5. You can check the size of `csi.dat`: `ls -l csi.dat`. It better be non-zero!
6. Load the file with Matlab or Octave as suggested above.

If this does not work, let us know where it goes wrong.

hmlsi, 2015-06-11:
Thanks a lot for generous replies @dpward  and @dhalperi . I'll surely let you know

hmlsi, 2015-06-12:
I am following your instructions carefully. I started to collect CSI data, but this time nothing was printed as last time (received 93 bytes: id: 26 val: 1 seq: 0 clen: 93). After checking size of csi.dat file, I came to know that its empty. So, I tried to read my last day's collected data. Its non-empty file (13490 size).
After running command 

csi_trace = read_bf_file('csi1.dat')

I am getting the error: error: 'read_bf_file' undefined near line 1 column 13
csi1.dat is my non-empty file

dhalperi, 2015-06-12:
It cannot find the script `read_bf_file.m`. Are you in the right directory for MATLAB to see the code?

dhalperi, 2015-06-12:
(It is probably not the same directory as `csi.dat`!)

hmlsi, 2015-06-15:
Sorry, I just overlooked that thing. csi.dat was in root directory and read_bf_file.m was in linux-80211n-csitool-supplementary/matlab.
Thank you for help

hmlsi, 2015-06-15:
one thing is unclear from my last question. Whether I have to unload and reload driver with csi logging every time or its a one time step? If its a one time step, why I needed to do it again to collect CSI data.
Much appreciation for your valuable responses and time

dpward, 2015-06-15:
The default value for the `connector_log` parameter in the modified iwlwifi module is 0x0.
* When you load the iwlwifi module manually with `modprobe`, you can set the value of `connector_log` directly on the command line; it will only set the value that one time. If the module is reloaded (for instance after rebooting) it will go back to the default value, unless you set the value again.
* To set the default value of `connector_log` to use when the iwlwifi module is loaded (either during boot, or using `modprobe` without any additional options), you can create a new file in the /etc/modprobe.d directory that contains the line: `options iwlwifi connector_log=0x1`

hmlsi, 2015-06-16:
Got it. Thanks

closed 2015-06-16 by dhalperi

jackyfff, 2015-09-20:
@hmlsi 

Hi, you've mentioned "I get the error "network is unreachable"". And you solved it by editing "/etc/network/interfaces". How do you do that?

I was facing the same problem. (While the router is working perfectly)

Thanks,

hmlsi, 2015-09-21:
I edited it by adding these lines:
auto wlan0
sudo iface
iface wlan0 inet dhcp
Since my router was setting ip using dhcp

tahmidzbr, 2016-09-12:
Hi @hmlsi can you please elaborate what you mean? Like what to type in the terminal? Im new to Linux please help!

niuqun, 2016-09-23:
@hmlsi Hello! I came into similar situations with you. How to edit /etc/network/interfaces so that you can ping the router that your client connects?

linkzhao, 2016-11-07:
I want to know how to process the CSI data  with C script?

fanzhangsuper, 2017-03-21:
@hmlsi Hi, I saw your response to disable network-manager, connect the AP and logging the data, here I connected to the AP, ping works, but the logging data command do not have any print out after I ping AP for more then 10mins. May you please let me know if you had the same problem before?

SevenOnes, 2017-08-22:
Hello everyone! I just wanted to know where could I have gone wrong if my pinging works but CSI is not being logged to csi.dat?

rt6n1, 2018-04-15:
Hello everyone! If I may refer to an older answer by  @dpward and @dhalperi - my problem is as follows:
- installation went fine
- AP is configured as 802.11n only, 2.4 GHz, 40 MHz BW, no security
- I connect to it in HT mode
- logging command produces a .dat file
- pinging goes OK
- but no messages are being displayed in the logging terminal window, and the .dat file remains zero-size.

Can you please help?

iftolito, 2020-02-04:
> Update : now that problem is solved by editing /etc/network/interfaces. I can ping to my AP now. In instructions, there is written that
> "After a few dozen pings, the rate selection should automatically switch to HT (802.11n) bitrates. At this point, log_to_file will start recording packets and the iwlwifi driver will print messages in the kernel syslog."
> what type of message will be?dozen of pings mean I have to ping few dozens of time explicitly or just receive and transmit dozens of packet? When I run "sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat" and ping my AP at different terminal, I have to terminate logging of CSI to a file. Is it normal to terminate it (Ctrl+C) or it should be closed automatically?
> 
> Later, I ping 60 times using
> 
> n=0
> sidra@sidra-Inspiron-N5110:~$ while (($n <= 60)); do ping -c 4 192.168.0.1; ((num++))
> 
> > done
> 
> but no message is printed on terminal as it should be according to instructions.

Hi i have the same problem like you, Problem: If I look on router side, router statistics show that IP address 0.0.0.0 is being assigned to my device. If I connect any other device, router is assigning IP address correctly. What can be the possible reason? when I try to ping my router, I get the error "network is unreachable". 
Can i know what do you have edit /etc/network/interfaces
because i dont find this folder thanks