dhalperi/linux-80211n-csitool-supplementary #46: how can i set the pc(install the csi tool) to ap model?

shenzhenyuyuyu, 2015-06-02:
I have tired many methods to set the pc to ap model, i want to connect mobilephone to the pc. So i can collect the csi between them. But i failed, is there any instructions to show how to set the pc(install csi tool)  to ap model?

dpward, 2015-06-02:
After you have followed the [CSI Tool installation instructions](http://dhalperi.github.io/linux-80211n-csitool/installation.html), you can install and configure *hostapd* to operate the Intel adapter in AP mode. There are more details about configuring *hostapd* here:
* https://wireless.wiki.kernel.org/en/users/documentation/hostapd
* http://w1.fi/hostapd/

Please make sure to see the [FAQ](http://dhalperi.github.io/linux-80211n-csitool/faq.html) first, especially 10 and 13.

There are some [sample *hostapd* configurations](https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/hostap-config-files) in the CSI Tool supplementary material. However please keep in mind that AP mode is not something that is specific to CSI Tool, and there are other online resources available to help you configure *hostapd* (unless you believe there is an issue that is specific to the modified driver for the Intel adapter).

dhalperi, 2015-06-17:
Closing for inactivity.

closed 2015-06-17 by dhalperi