dhalperi/linux-80211n-csitool-supplementary #48: db

TadiGuy, 2015-06-06:


dpward, 2015-06-06:
Not only is that the correct conversion, I get the same plot as you using MATLAB R2015a on 64-bit Linux &mdash everything is shifted down by 26 dB.

@dhalperi I assume this is the result of some change to the MATLAB scripts after that plot was added to the FAQ. (I've only made one minor change to the MEX-file that wouldn't affect this.) Is this new result correct and should we just update the plot, or is this worth looking into further?

dpward, 2015-06-06:
@dhalperi The commit that caused the plot to change is:
02af39e matlab: add quantization error to get_scaled_csi

dhalperi, 2015-06-06:
Sure, that makes sense; the quantization error affects mostly very strong links, and I'm sure I didn't update the image after that change. Feel free to submit a new image.

dpward, 2015-06-06:
What about this text in the FAQ below that plot:

> [...] we can see that all subcarriers are stronger than 40 dB, so the frequency-selective fading should cause few errors even using the highest rate. There is so little error, in fact, that MATLAB's error functions can't distinguish it from zero, and the SNR is effectively infinite.

Now the lower bound is actually 15 dB instead of 40 dB. I don't think that statement is true any more, right? (I'm actually referring to the plots in [your dissertation](http://arxiv.org/pdf/1301.6644.pdf) here.)

closed 2015-06-06 by dpward

dhalperi, 2015-06-06:
Yeah that text should probably get updated. I'm on mobile right now, but can take a look later. (Also feel free to update the text yourself...)

Thanks, I appreciate it.

dpward, 2015-06-06:
@TadiGuy Also for future reference, the [signal package for Octave](http://wiki.octave.org/Signal_package) contains the db() function that you were missing.

dpward, 2015-06-06:
Sorry I totally misread that page &mdash; which actually says that the db() function is *not* yet implemented in Octave.

TadiGuy, 2015-06-07:
Thanks a lot.