dhalperi/linux-80211n-csitool-supplementary #51: the same sub carrier's data sample varies significant, why

xuekevin, 2015-06-10:
One Location I will get 60 samples, I found that the data I got is like following
[image]
Is it normal? thanks!

dhalperi, 2015-06-10:
In general: Yes it is normal, especially as the transmitter changes its antenna selection or other rate configuration, or if there is movement in the area during the study, or if there are long periods of time between measurements.

You did not give us enough information to answer questions about your specific configuration.
- what is the transmitter device?
- how is the network configured? (AP, injection, ad-hoc?)
- did you set rates manually or are you using rate selection?
- how are you extracting the vector you show in the above screenshot?
- for your analysis, if the transmitter is an intel5300 device, are you using `get_scaled_csi_sm`?

dhalperi, 2015-06-16:
Closing because of inactivity. Let me know if you have updates.

closed 2015-06-16 by dhalperi