dhalperi/linux-80211n-csitool-supplementary #52: In the same location at different time to get three very different data？

chenyongxi, 2015-06-10:
Collecting data at different times at the same position（for example half an hour interval）.The data I collected very different. One's mean（30 subcarriers） is 22db， the other's mean is 15db.This is normal? Where was I wrong?

dhalperi, 2015-06-10:
This is normal.

1. Different transmit rates (# of streams, transmit antenna selection) affect the link.
2. Movement in the environment or repositioning any RF-visible materials, such as metal water and glass, including people, will affect the link.
3. Even, sometimes, rebooting the hardware will result in recalibration which can affect the reported CSI.

closed 2015-06-10 by dhalperi

chenyongxi, 2015-06-10:
Thank you！The second case I have taken into account as far as possible avoid, the measurement at night.  I want to know how to avoid the third case.( rebooting the hardware will result in recalibration which can affect the reported CSI.)

chenyongxi, 2015-06-10:
Such lager fluctuations is nomal?

dhalperi, 2015-06-10:
Yes. Number 1 is the most likely still, however, if you've ruled out any movement or change in the environment.

chenyongxi, 2015-06-10:
But measurement is the same ap.

chenyongxi, 2015-06-10:
Throughout the experiment with a same device I have used.  Are there any ways to avoid the impact of  the first case (Different transmit rates (# of streams, transmit antenna selection) affect the link.)?

dhalperi, 2015-06-10:
It sounds like you are connecting to a commercial AP? The ap will select
rates and or antennas, and that will be out of your control. You can
compensate for this by clustering your received csi reports based on the ap
configuration. You can observe the transit rate directly; you will have to
come up with some way to detect when the ap uses antenna selection or beam
forming.

Alternately, use both transmitter and receiver as Intel 5300 devices and
use injection mode, then you will precisely control the transmitters use of
antennas and rates, etc.

chenyongxi, 2015-06-10:
I did connecting to a commercial AP. We can not connecting to a commercial AP?

dhalperi, 2015-06-10:
You absolutely may connect to a commercial AP – as you have already found, the CSI tool will still work!

However, you have no way of guaranteeing that the commercial AP will not do things like change which antenna it sends data from, for instance. Thus you have no reasonable expectation that the CSI will remain stable over time as the transmitter changes its behavior in unknown ways!

You may find it useful to read my thesis, especially chapter 4: http://homes.cs.washington.edu/~dhalperi/pubs/halperin_thesis.pdf

chenyongxi, 2015-06-11:
If I want to avoid receiving CSI such a large jump, how can I do?

chenyongxi, 2015-06-11:
Have any commercial AP sends data stable？

dhalperi, 2015-06-11:
Not that I know of. The recommended strategy for these types of long-term
measurements is to use a CSI Tool transmitter as well, so that you have
full control over both sides. Any commercial device is going to have
unknown behavior.

chenyongxi, 2015-06-11:
I don't understand what is you said these types of long-term measurements is to use a CSI Tool transmitter. Can you say that more points？

dhalperi, 2015-06-11:
Use an Intel 5300 device running the CSI tool to both transmit and receive.
Do not use a commercial AP.