dhalperi/linux-80211n-csitool-supplementary #55: How to forbid the intel 5300 to work on 5GHz?

gitbub1992, 2015-06-12:
Dear Daniel：
       I have many APs with the same essid (different MAC address)，while half of that work on 2.4Ghz,another half work on 5Ghz.My problem is that my laptop with Intel 5300 always connect to the 5Ghz APs and can not connect to the 2.4Ghz ones,even I have assigned the MAC address.So how can   I forbid the 5300 work on 5Ghz?
     Thanks in advance.

gitbub1992, 2015-06-12:
@dpward

gitbub1992, 2015-06-12:
My problem have been solved after I changed the para of iwlwifi mod.So you can close this issue,thank you all the same.

dhalperi, 2015-06-12:
Great!

closed 2015-06-12 by dhalperi

gitbub1992, 2015-06-13:
@dhalperi,new problem arises,I can not get CSI  anymore using log_to_file after I changed the para of iwlwifi .(the iwlwifi needed to be remod and reload),Will recompiling the iwlwifi and setting 5ghz_disable=1 work?Thanks you very much!

dhalperi, 2015-06-13:
no idea, you're in uncharted territory here! Try it and let us know how it goes :)

gitbub1992, 2015-06-13:
Thank you for your quick reply!!BTW, Where can I get The subcarrier index of CSI in 5Ghz?Since the total number of 5Ghz is 108.

dhalperi, 2015-06-13:
I think you have some confusion – the number of subcarriers is not band-dependent. You can refer to the IEEE 802.11 standard for more information.... https://standards.ieee.org/getieee802/download/802.11-2012.pdf

gitbub1992, 2015-06-13:
Sorry,I didn't correctly express my problems.My question is that  the corresponding subcarrier index sequence of  the 30 CFR exported by CSITools  in 5Ghz is the same with that of 2,4Ghz,which is [-28，-26，…，-4，-1，1，3，…，25，27，28]?

gitbub1992, 2015-06-13:
The problem finally was solved by setting 5ghz_disable=1 in the file iwlwifi.conf under /etc/modprobe.d.If no file exists, create a new file,and Add a new line:options iwlwifi 5ghz_disable=1.
The refrence ：http://askubuntu.com/questions/419867/qualcomm-atheros-ar9485-wireless-network-adapter-not-working-on-ubuntu-13-10

dhalperi, 2015-06-13:
Thanks for the update!

Yes, the sequence of subcarriers is the same for 20MHz channels independent of band.

gitbub1992, 2015-06-14:
Much appreciation for your kind reply!!