dhalperi/linux-80211n-csitool-supplementary #56: when i use ifft( inverse Fourier transform)  transform csi from frequency domain  to time domain?

shenzhenyuyuyu, 2015-06-15:
Like the title said, the distance between two adjacent subcarraier is  c/20Mhz= 15m  (when is in 40Mhz condition, distance is c/40Mhz=7.5m),  is that right?   I have a question,  when the subcarries is 30, the distance is 15m, but when the subcarries is 56(full subcarries) the distance is also 15m?  Can someone give me an answer.

dhalperi, 2015-06-16:
802.11 subcarriers are 312.5 kHz apart (64 subcarriers in 20 MHz); you can learn more in the 802.11n standard available at <http://standards.ieee.org/about/get/802/802.11.html>.

Please also read the FAQ about which subcarriers are measured in the CSI -- there are only 30 measurements, but that does *not* mean that the hardware uses only 30 subcarriers to transmit data.

closed 2015-06-16 by dhalperi

star90, 2015-12-03:
I have seen somewhere that there are 56 subcarriers in 20MHz...

dhalperi, 2015-12-09:
There are 56 data subcarriers in 20 MHz (64 total), but only 30 are measured. Please read the FAQ.

shaozhuchuang, 2016-06-08:
@dhalperi @dpward  if i use HT20, what is the frequency spacing between two consecutive subcarriers? 20 MHz/64 or 20 MHz/56 or 20 MHz/30? Thank you!