dhalperi/linux-80211n-csitool-supplementary #57: Received 93 bytes?

chenyongxi, 2015-06-16:
why I offen print " received 93 bytes : ............" when it only 93 byte the csi is 1*1*3.  When received 213 byte the csi is 1*3*3, but only a little time I can receive 213 byte, why?

dhalperi, 2015-06-16:
What do you mean by "1x1x3"? Do you mean 1x1x30 – a single-transmit-single-receive CSI measurement for all 30 subcarriers? That is totally normal, depending on your hardware and link configuration.

It sounds like either your wireless link is weak or your receive antennas do not work well (wrong frequency band, for example)?

What are the RSSIs reported for the three receive antennas?

chenyongxi, 2015-06-17:
Really a single-transmit，But two Antenna receive。 sometime receive 1x2x30(153byte), but more time receive 1x1x30(93byte).，I can't control it.   I don't know how I can see  antennas are work well? 
RSSI sometimes Antenna a and Antenna  c is 0 and b has value, sometimes Antenna b and Antenna  c is 0 and a has value, the received Antenna I connect a and b.

chenyongxi, 2015-06-17:
30 subcarriers is normal

dhalperi, 2015-06-17:
I would suspect that something is wrong with your antenna configuration. Make sure that you've connected antennas to all 3 ports and that the antennas you use are "dual-band" antennas rated for both 2.4GHz and 5GHz operation.

closed 2015-06-17 by dhalperi

chenyongxi, 2015-06-17:
But I have to use this computer to get over data with ubuntu 10.04 http://dhalperi.github.io/linux-80211n-csitool/old_installation.html.  But now I want use ubuntu 12.04 experiencing these problems.

dpward, 2015-06-17:
Since Ubuntu 12.04 uses Linux kernel 3.5, the driver should actually be unchanged from the old instructions. The firmware is the very same as well.

If you are actually running a point release (e.g. 12.04.4), which has a later kernel version, and you want to test to be sure, try Ubuntu 12.04 instead, and let me know if it behaves differently. However it seems much more likely that this is not the issue.

chenyongxi, 2015-06-17:
how to the kernel enforces module signature verification?

dhalperi, 2015-06-18:
http://dhalperi.github.io/linux-80211n-csitool/faq.html

BaoYu0721, 2015-10-26:
Excuse me... I come up with almost the same problem.. At most time, I can only receive 93 bytes, namely, it goes into the 1 * 1 * 30 branch..  And sometimes, I can also receive 153 bytes, namely, it only detect 2 rx, however, it is a wrong branch.....
As for the value of rssi, I got a result like this: rssiC is always 0, either rssiA or rssiB has a value around 40, namely, at the same time, rssiA is around 40 and rssiB is 0, or , rssiA is 0 and rssiB is around 40...

I am confused about this, please help...... Thank you very much!

dhalperi, 2015-10-27:
This sounds like an antenna problem with your hardware.

lixiaohai, 2015-11-06:
hi,dpward!

I come up with the same problem.AT first, i use ubuntu12.04.4,which has a later kernel version(3.13.0-61.100)as you sayed,so the received bytes changed from 93 to 213.Then i changerd my version to ubuntu12.04.2(kernel is 3.5),and the result becomes normal(213 bytes).So,i think that's just the issue.

ps.There is no antenna problem with my hardware.

dpward, 2015-11-23:
Hi all, my apologies that I've been very inactive here lately. Let me see if I can reproduce this over the next few days.

junsujang, 2017-08-04:
Hi all,

I am facing a similar problem.

I am using 1x3 SIMO, where I initially receive 1x3x30 but I end up receiving 1x1x30 every time. I am also using kernel version3.13.0-32-generic. Is the kernel version the issue?