dhalperi/linux-80211n-csitool-supplementary #58: auto disconnect while logging

jinzhang-unsw, 2015-06-16:
I use commercial router as ap and laptop for logging csi. Router has 3 tx antenna and I set ``0x1c916`` as transmission rate and work under 5ghz. The laptop firstly log csi very fine. All 3 * 3 csi data have been recorded. But after some moments the csitool can only log part of csi data, the recorded packets size drop from 573 to 393 and down to 213 and disconnect in the end. Every time after reloading csi tool it works fine but then auto disconnections happen very often. Do you know what the reason? Is there any major changes or updates since your new csi tool launch? I download it very early. Thank you.

dhalperi, 2015-06-16:
The size of the recorded CSI measurements varies based on how many transmit streams and receive antennas are measured. Usually a varying packet size means that the transmitter is doing rate selection.

What type of experiments are you running – are you moving around, for instance? Is the link very weak? Etc.

lidongmeng, 2015-06-17:
@jinzhang-unsw hello, how can I set the router's transmission rate to a specific number? Thank u!

dpward, 2015-06-17:
@lidongmeng You can't set the transmission rate on a commercial access point.

You can only set the transmission rate with the Intel adapter. So if you want to use access point mode and control the transmission rate from the access point, you need to set up a station with an Intel adapter operating as an access point.

lidongmeng, 2015-06-17:
@dpward ok, thank u.

jinzhang-unsw, 2015-06-17:
@dhalperi   I've guessed so. I place laptop and router very close 2meters maybe and remain static no one move around devices. I have set the tx rate and it shows the laptop work under the defined rate at first but later drop to even auto disconnection, because the packet size drop accordingly. Is it due to the environment interference? Is there any way to maintain these kind of communication performance?

jinzhang-unsw, 2015-06-17:
@dpward I set router as access point and laptop connect with the router. In the laptop end, I run FAQ.1.C Setting the transmit rate to another station. Is this the right way to set the tx rate?

dpward, 2015-06-17:
@jinzhang-unsw That will set the transmit rate from the laptop to the AP. However it doesn't set the transmit rate in the other direction (from the AP to the laptop), and those are the packets for which CSI is collected.

jinzhang-unsw, 2015-06-17:
@dpward Ah this may be the reason. So only the two laptops can control the tx rate while logging CSI?

dpward, 2015-06-17:
@jinzhang-unsw Yes if you have two laptops communicating directly, you can control the transmit rate in both directions while logging CSI. You need to configure it in a mode so that they are communicating directly (see FAQ 5):
* one laptop as an AP, and the other as a client; or
* both laptops in ad-hoc mode; or
* both laptops in monitor mode and using the injection code

dhalperi, 2015-06-17:
@jinzhang-unsw even worse, your AP may not support the three-stream rate you set. In that case, the AP may not receive your packet acknowledgements, which might explain why it lowers the rate and eventually disconnects. Try using a rate like 0x8102 (only one TX stream, but 19.5Mbps) and see if that keeps the AP's transmit rate (i.e. the rate of your received packets) going.

However, for truly controlled experiments you will need to control both devices as @dpward says above. I *highly* recommend using two laptops with the CSI Tool and injection mode.

closed 2015-06-17 by dhalperi