dhalperi/linux-80211n-csitool-supplementary #59: Need help on injection mode!!

lidongmeng, 2015-06-17:
First, I install the libpcap-dev, then I download LORCONv1, but I do not know how to compile it, just use make? but it does not has makefile? Help me plz. Thank u!

dpward, 2015-06-17:
You need to run the configure script first:

```
./configure
make
sudo make install
```

lidongmeng, 2015-06-17:
@dpward Thank u. I install the LPRCONv1, then cd the injection fold to run "make" to install the injection. The transmitter and the receiver should do the same process? Next, when I run the sh in the transmitter and receiver separately, should the receiver connect to the transmitter?

dpward, 2015-06-17:
@lidongmeng Yes, you will use the same steps to set up the transmitter and receiver (install libpcap-dev, build and install LORCONv1, and build the injection code).

This will not create a "connection" like the other modes; you will not have an IP address on the wireless interface and you can't use it like a regular Wi-Fi network anymore. Instead, you can "inject" individual 802.11 frames that have a special source and destination address, and other stations will be able to collect CSI for these frames if they are receiving.

lidongmeng, 2015-06-17:
@dpward I don't really understand "you can "inject" individual 802.11 frames that have a special source and destination address", the transmitter send packets, but there is no connection between the transmitter and the receiver, can the receiver get the packet? what is the address of the packets the transmitter send?

dpward, 2015-06-17:
@lidongmeng See FAQ 5: "The packet is addressed both to and from a hardcoded, fixed address 00:16:ea:12:34:56."

lidongmeng, 2015-06-17:
@dpward In other word, the transmitter send packets to the hardcoded , and the receiver monitor the address, when the packet get, receiver get the packet and log it? am I right?

dpward, 2015-06-17:
@lidongmeng Yes that's right.

lidongmeng, 2015-06-17:
@dpward Does the length of the packet affect? Normally, which length should I set, 100??

dhalperi, 2015-06-18:
@lidongmeng how do you think the length of the packet affects packet reception? :)

Use the configuration in which things are most likely to succeed!

dhalperi, 2015-06-27:
Closing for inactivity.

closed 2015-06-27 by dhalperi