dhalperi/linux-80211n-csitool-supplementary #60: On the training sequence in 802.11n

linkin8834, 2015-06-20:
Hello,
     As far as I know, 802.11n uses the **two long preambles** to get the estimated CSI. I am wondering, could I say that the CSIs I captured using Intel 5300 can be grouped into segments composed by two consecutive CSIs belonging to the training sequence of one package?  For instance, if I capture
      CSI1  | CSI2 | CSI3 | CSI4 ...
   
    Can I conclude that, CSI1 and CSI2 are from the 1st package, and CSI3 , CSI4 are from the 2nd package? If so, I could use some algorithms to estimate/compensate the frequency offset.
     Thanks!

linkin8834, 2015-06-20:
Or, are the CSIs obtained from the HT-LTF? For example, if I use 2 spatial streams, does it mean that the HT-LTF would look like this:
[image]
and we can get a total of 4 CSIs (2x2)? Thanks!

dhalperi, 2015-06-20:
You receive at most one CSI measurement per received packet. I am no longer
familiar with the exact process, but the CSI is estimated and refined
during the entire preamble.

linkin8834, 2015-06-21:
Thanks!

    Another question is: is it possible to tune the txpower? I am in the regulation domain of US. It seems that the 5300 txpower is fixed as 15dBm. I tried iwconfig wlan0 txpower 20 but failed. I want to adjust it for educational purpose (to see the difference of CSIs with an increasing power). Some solutions said I should change the reg domain to BO. It does not work for me. 

    Thanks!

linkin8834, 2015-06-21:
Hmm, with connector_log=0x5 I should be able to get the package payload as well. I see entries like 
```
Entry size=30, code=0xC1
```
in the log file. Does it say anything to the sequence number or package payload? 
Thanks, and sorry for asking so many **silly** problems all the time. Your tool really rocks (and the academic checking tool) :+1:

dpward, 2015-06-21:
Regarding the transmit power, see FAQ 13. The EEPROM in the Intel adapter contains regulatory settings, and those settings get combined with the regulatory domain settings that you have selected in the Linux kernel. The settings written by Intel seem to be more restrictive than the regulations in the United States.

linkin8834, 2015-06-21:
Hello David,
        I tried to reduce the txpower and it works. It seems that changing domain tricks does not work for intel...anyway, it is good that at least I can reduce it, so that I can study its effect on performance. Thanks!

linkin8834, 2015-06-22:
For the logging problem, I think it is the cmsg->seq. I commented it out from the log_to_file.c and I missed the sequence number...

Thanks!

dhalperi, 2015-06-22:
Please read FAQ#12 for information about the 802.11 sequence number. The `cmsg->seq` is actually unused, but signifies the sequence number for the socket by which the kernel sends messages to userspace.

linkin8834, 2015-06-22:
Hi Daniel,
    I turned the connector_log id to 0x5 and I see that some packages are in different size with the 3x3 package (573 Bytes). I guess that they correspond to the additional information of the package (not just the beamforming information). I think log_to_file dumps everything to the output file. Should I change the read_bfee.c to dump the payload into workspace (or should I just use wireshark). Thanks!

dhalperi, 2015-06-22:
`read_bfee` produces only CSI measurements. If you want the packet payloads, you need to write code to read them.

dhalperi, 2015-06-22:
573 bytes is the size of a 3x3 CSI measurement. The payloads are in different messages which you will find in MATLAB.

closed 2015-06-22 by dhalperi

linkin8834, 2015-06-22:
Dear Daniel,
     Got it. Thanks so much :}

linkin8834, 2015-06-22:
Hello,
        Changing txpower only works with the **original firmware**. Given the modified one, whatever I change the txpower, it always stay at 15 dBm. Does the modified firmware prohibit txpower change? Thanks!

dhalperi, 2015-06-22:
Please create separate issues for separate questions.

swashah, 2016-11-22:
@linkin8834 : Hi, I want a sequence number with every packet while injecting packets and receiving them on two laptops in monitor mode. Subsequently I intend to use this seq no. for synchronizing the data collected by two laptops. I will appreciate your help. Thanks