dhalperi/linux-80211n-csitool-supplementary #61: On changing txpower using modified firmware

linkin8834, 2015-06-22:
Hello,

Changing txpower only works with the original firmware. I can tune it from 0 dBm all the way to 15 dBm. However, given the modified one, whatever I change the txpower, it always stay at 15 dBm. Does the modified firmware prohibit txpower change? Thanks!

dhalperi, 2015-06-25:
Yes, it is possible to change the tx power with the new firmware; this was a key part of my thesis.

closed 2015-06-25 by dhalperi

DontKnowMy, 2017-03-31:
Which one is the new firmware? And how can I change the txpower?

DontKnowMy, 2017-04-05:
@dhalperi  could you please state how to lower the txpower using what new firmware? would be very appreciated. Thanks

linkin8834, 2017-06-29:
@DontKnowMy No. With the firmware you are unable to set the transmit power. However, there are still many ways to do the experiment to tune the txpower:

1. Connect RF attenuators to the 5300 antennas on the TX side. If you want to control the receive sensitivity, connect RF attenuators to the 5300 antennas on the RX side.

2. Use another laptop with 5300 (or any other WiFi cards) in AP mode in Linux. Do not use modified firmware on that laptop, though. Then, just use iwconfig to change the txpower. I am able to tune to txpower to 0dBm with ease.

Statisticss, 2018-06-01:
@dhalperi I read your thesis carefully and I found that in chapter 5.3, under 'Transmit Power Control', you said: 'I modified the driver and firmware to enable transmit power variation. With these changes, I can vary the transmit power level from -10dBm (100 uW) to +16dBm (40mW) in steps of 0.5 dB.' Could you plz give us some suggestion on how to do that? Thanks!