dhalperi/linux-80211n-csitool-supplementary #63: Antenna C doesn't work well?

co12, 2015-06-24:
Dear dhalperi,

I use the ubuntu 12.04 LTS with the kernel 3.2.0-23-generic, the laptop equipped with 5300 link connects to an AP router with three antennas supporting 450Mbps. the laptop is receiver and the AP is transmitter.

But when I using the three rx antennas, "Nrx" in the CSI information is always 2 in most cases, rssi_a and rssi_b are perfect as your data in FAQ#2, but rssi_c is 0. When Nrx is 3, the rssi_c is belows 20.

So I guess there is some problems with the antenna C, so I set SISO only using antenna C to receive,
echo 0x4 | sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/debug/rx_chains_msk

the result shows that rssi_c is about 40 as normal as in your data, I have no idea of the reason,
by the way, the "Ntx" never comes to 3, and I also has the issue#57, why? 
 please help, thank you!

dhalperi, 2015-06-27:
How are the antennas configured physically? If they are not spatially diverse or have crosstalk, Intel's receiver hardware can sometimes disable one. Antenna C is usually the most susceptible because it's in the middle

co12, 2015-07-01:
Thanks for your answer! the meaning you guess is that the antennas in the 5300 wireless card are not manufactured well together? How can I detect it? Need I disassemble the laptop thinkpad X200?

dhalperi, 2015-07-01:
You should not need to fully disassemble the laptop, but you do need to confirm that, for example, all the antennas are present and attached!

I found a web forum where people are complaining that Lenovo only attached 2 antennas to the NIC, in which case Antenna C (the most likely left unattached) will indeed perform terribly! 

http://forum.notebookreview.com/threads/x200-with-intel-5300-wireless-card-question.519296/

co12, 2015-07-01:
OK，I will confirm that later. Much appreciate for your help！

co12, 2015-07-02:
Unfortunately I find that the three antennas are attached very well, it doesn't look like this reason.

I suspect that whether it has something to do with the tramsimitter of TP-LINK or not? 

or it is said that the NIC will select the antenna C to serve the data communicaiton when using three antennas to receive?

Most noteworthy is that when I only active the antenna C to receive the packet, the value of rssi_c is normal.

dhalperi, 2015-07-02:
What channel are you using? I have also found that vendors may attach a 2.4GHz-only antenna to port C, so if you are using a 5 GHz channel then you may be seeing this effect.

How much weaker than rssi_a and rssi_b is rssi_c?

dhalperi, 2015-07-02:
Also, can you confirm that bluetooth is fully disabled on this laptop?

co12, 2015-07-03:
I am using the kernel 3.7.5-csitool in ubuntu 10.04, the channel is in 2.4GHz frequency band because the AP is 2.4GHz band.

I have already disabled the bluetooth, but it doesn't work.

the rssi_c is usually 10 or so, the rssi_a and rssi_b are usually about 40.

dhalperi, 2015-07-03:
This makes sense to me – when antenna C is 30 dB weaker than antenna A and
B, the 5300's DSP may well just drop its signal. Information-theoretically,
it provides very little benefit in terms of aggregate (effective) snr, and
it uses lots of energy to process that data.

co12, 2015-07-03:
I think the wrong thing is the antenna C, because when I change the connection mode as follows:

antenna C(white line)<---> TR1, antenna A(gray line)<---->TR3, antenna B(black line)<----->TR2
（original mode: A<---->TR1, C<------>TR3, B<------>TR2）

the rssi_a is about 10, the rssi_b and rssi_c is normal.

Is that mean that antenna C is damaged?

dhalperi, 2015-07-03:
Either damaged, poorly placed in the laptop, or not good for the channel or
band you're using.

co12, 2015-07-07:
The problem is solved.

when I change to another X200, the issue persists, but when I change to X301, the issue is gone.

So it is the problem of X200 itself, I closed the issue.

closed 2015-07-07 by co12