dhalperi/linux-80211n-csitool-supplementary #65: Problems connecting from android phone

zeroGMac, 2015-06-25:
I'm trying to measure CSI for connections between an Android based smart phone and the intel adapter. The problem is that the phone disconnects after a few seconds after connecting to the intel adapter in AP mode. Then re-connects, stays for a 4-5s and then again disconnects.

Also I've noted that when using the internet from the phone, while connected, the connection remains open for as long as I keep using the internet. (i.e it seems like the phone loses connection when there is no network activity - And I've switched off 'wi-fi off during sleep' and 'wi-fi optimization' on the phone.)

When connecting from a laptop I'm not facing any such issues. Can this be related to the modified driver/firmware ? Any pointers on what can I do to resolve this? 
(I have the adb logs when the phone gets connected/disconnected, but I couldn't discern any useful information from it. Let me know if you want me to post it here.)

dpward, 2015-06-25:
I would suggest looking in the hostapd log as well &mdash; it may have clues.

One way to distinguish CSI between different devices is for the driver to log received MPDU payloads in addition to the beamforming information. To do this, set `connector_log` to 0x5 (which is `IWL_CONN_BFEE_NOTIF_MSK | IWL_CONN_RX_MPDU_MSK`). The payload will appear before the CSI sample and will contain the 802.11 header. If you are using the MATLAB/Octave code to parse the connector log it will require modification to parse the MPDU messages and extract the address from the 802.11 header, for example.

zeroGMac, 2015-06-25:
I had tried using 0x5 flag. But it seemed to have returned the same information in the CSI log although more packets where being logged. I was getting this information by stepping through the lines extracting the information in log_to_file.c. I might be missing something. I'll try both the suggestions. Thanks.

zeroGMac, 2015-06-25:
And no luck with hostapd.log:

Configuration file: /etc/ap-hotspot.conf
Failed to update rate sets in kernel module
Using interface wlan0 with hwaddr ... and ssid 'apnew'
wlan0: AP-STA-CONNECTED ...

dpward, 2015-06-25:
Can you try increasing the log level in the hostapd configuration?

When parsing the connector log, the MATLAB/Octave code will skip over any messages that are not beamforming information messages (0xbb). You will have to add to code to process the received MPDU messages (0xc1).

zeroGMac, 2015-06-25:
The current log level is :
 logger_syslog=-1
logger_syslog_level=0
logger_stdout=-1
logger_stdout_level=0

(From the sample config file supplied with tool) -- And it says 0 is the highest.

I still get the same messages in the log file:

Configuration file: /etc/ap-hotspot.conf
Failed to update rate sets in kernel module
Using interface wlan0 with hwaddr 00:21:6a:c8:fe:ac and ssid 'apnew'
wlan0: AP-STA-CONNECTED 1c:62:b8:a8:71:15

zeroGMac, 2015-06-25:
And I get this when I checked syslog:
hostapd: wlan0: STA ... IEEE 802.11: disassociated due to inactivity
hostapd: wlan0: STA ... MLME: MLME-DISASSOCIATE.indication(..., 4)
hostapd: wlan0: STA ... MLME: MLME-DELETEKEYS.request(...)
hostapd: wlan0: STA ... IEEE 802.11: deauthenticated due to inactivity
hostapd: wlan0: STA ... MLME: MLME-DEAUTHENTICATE.indication(..., 2)
hostapd: wlan0: STA ... MLME: MLME-DELETEKEYS.request(...)

dpward, 2015-06-25:
You can find similar issues with other chipsets using hostapd: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=659059

I'm not sure this issue is specific to the CSI Tool driver or firmware; to narrow down the issue, you probably need to obtain a different wireless adapter that also supports AP mode, and try it in place of the Intel adapter to see if you get similar behavior. If the cause is the CSI Tool, I'm not sure we have the ability to do anything about it unfortunately (see FAQ 10 and FAQ 9).

zeroGMac, 2015-06-25:
Okey. Thanks for the help. I've made some progress.

zeroGMac, 2015-06-30:
Is there a place where I can find some sort of documentation on MPDU fields and their relation w.r.t the read in bytes.

dpward, 2015-07-01:
You can refer to section 8.3.2.1 in the [IEEE 802.11-2012 standard](http://standards.ieee.org/getieee802/download/802.11-2012.pdf) for the MAC data frame format (Figure 8-30) and address field contents (Table 8-19).

dhalperi, 2015-07-14:
Closing due to idleness. Please reopen if you still have issues.

closed 2015-07-14 by dhalperi

zeroGMac, 2015-08-26:
I have created a new c file very similar to read_bfee.c to read the mpdu packets. The code that I use is something like below for extracing out the address from the bytes passed in:
        
    long long addr1 = (((((((((inBytes[4]<<8)|
								inBytes[5])<<8)|
								inBytes[6])<<8)|
								inBytes[7])<<8)|
								inBytes[8])<<8)|
								inBytes[9];

is this correct? (I am using the MAC data frame format from the above mentioned section 8.3.2.1 of the document). 

Also when I use connector_log as 0x5 the the tool logs a lot of packets even before I connect to the hotspot using a device. Is this expected behaviour?  (However CSI is logged only after I connect a device and use the network as expected)

YanMengSJTU, 2016-08-02:
@rakeshbala 
Hi, excuse me, I also have the same problem as you. When I use hostapd to open a non-encrypted Wi-Fi, my mobile phone will disconnect the WiFi after few seconds. Perhaps you have sovled this problem, can you tell me how to solve this problem? Thanks a lot!

swashah, 2016-11-20:
@rakeshbala : HI, I want to synchronize CSI recorded by two laptops in monitor mode while thord laptop is injecting packets. Can you please suggest how you achieved this, and what modifications you made.