dhalperi/linux-80211n-csitool-supplementary #67: Why there is an uncertain delay at the start of measured CIR samples?

chenyongxi, 2015-06-26:


dhalperi, 2015-06-27:
Because the NIC doesn't compensate for it.

closed 2015-06-27 by dhalperi

chenyongxi, 2015-06-29:
I still do not know.Can you give me some more detailed description？Thanks！

chenyongxi, 2015-06-29:
I put my question detailed point.  First of all, I got the 30 sub-carrier frequency-domain information.  I got their temporal information processing by IFFT.  In theory, the peak in the time domain should delay at 50ns.  However, the largest peak in the time domain, mainly in 150ns or 200ns or 250ns or 300ns, which peaked at 200ns at the most.
I want you to give me detail about why this happens.The more detailed the better。Thanks！

dhalperi, 2015-06-29:
I don't know, sorry. There is a lot of processing on the NIC, not to mention data quality issues with the CSI and lots of other issues.

There are papers where the authors have looked at the delay profile of the CSI and used it for a variety of purposes. Please contact those authors.

ZhaoZifei, 2017-03-15:
@chenyongxi I have puzzle with the IFFT of CFR, in matlab if it is use the code below?
`cir = ifft(cfr,30); %%cfr is just the csi in one packet extract using the function like read_bf_file,get_scaled_csi`
and I don't know how to get the delay time in x-axis.
Thanks in advance, hoping for your reply.