dhalperi/linux-80211n-csitool-supplementary #68: on 40MHz 5G bandwidth

linkin8834, 2015-06-26:
Hi David and Daniel,
     I am trying to configure the 5300 to run under 40MHz mode in 5G band. I use channel 64 (5.32GHz) with the option HT40-. However, the rate field in the csi trace is still 275. It seems to me that it should be something like 2048 (as I found in the function remove_sm.m for 40MHz mode spatial mapping). 
     On the other hand, how to confirm if the 5300 is working under 40MHz? In iwconfig I cannot see the BW. In wavemon the same problem. Thanks!

linkin8834, 2015-06-26:
I think I got it. I forgot to reset the rate flag 0x1c113 to its 40MHz version. My bad...

dpward, 2015-06-26:
Ok great. Instead of `iwconfig` I would suggest you use the replacement command `iw`. It provides more control and exposes more information about the wireless interface.

linkin8834, 2015-06-26:
Hi David,
    Thanks for your suggestion :D

linkin8834, 2015-06-26:
Working confirmed. I will close this issue.

closed 2015-06-26 by linkin8834