dhalperi/linux-80211n-csitool-supplementary #69: How to get received WiFi beacons from AP?

xuekevin, 2015-07-01:
I read some paper, they proposed some system which disable active scanning, and only uses beacons from APs. I wonder how to get beacons message using your system? thanks!

dhalperi, 2015-07-01:
FAQ number 5. You can only measure CSI for beacons from APs to which the CSI Tool is an associated client. There is at most 1 AP at a time in this implementation.

closed 2015-07-01 by dhalperi

xuekevin, 2015-07-01:
Could you give me more tips ? I have read FAQ number 5, according to your answer, my understanding is I need to make the CSI Tool an associated client, what is associated client?   But how can I achieve that ?  What messages can I get from FAQ number 5 ?

dhalperi, 2015-07-01:
You have clearly not followed the setup and installation instructions.

xuekevin, 2015-07-02:
all right! I will review the instructions, thanks! :)