dhalperi/linux-80211n-csitool-supplementary #70: How to receive beacon packets from the AP?

xuekevin, 2015-07-02:
Sorry for ask similar question again, I think maybe I didn`t make my question clear. I can get CSI messages when I ping the AP which follows the instructions you have offered. Now I want to receive beacon packets from the AP, I learn that the AP will broadcast beacon packets periodical, How can I get these beacon by making some change ?. In other words, How can I get CSI message without ping? Thanks a lot!

dhalperi, 2015-07-02:
The AP needs to send the beacons at HT rates. Is it doing so?

xuekevin, 2015-07-02:
I am tring to find how to set the beacons at HT rates, If the AP send the beacons at HT rates, which means I excute log_to_file command then, I can receive CSI message and write in the file ?

dhalperi, 2015-07-02:
Please read FAQ.5, which has the answer.

linkin8834, 2015-07-03:
@xuekevin It seems that it cannot. Ping would be OK.

closed 2015-07-10 by xuekevin

tfsanam, 2017-04-07:
@dhalperi you said: The AP needs to send the beacons at HT rates. Is it doing so?
 Can you please tellmehow can we check that?