dhalperi/linux-80211n-csitool-supplementary #71: Antennas are not working fine

hmlsi, 2015-07-02:
Hi,
Whenever I try to collect CSI data, number of antennas in different received packets is different while keeping the environment same. Why this is so?

dhalperi, 2015-07-02:
Transmit or receive antennas?

Hardware setup?

What type of device is transmitting the packets?

What type of device is receiving the packets/measuring CSI and how is it configured?

hmlsi, 2015-07-02:
1.Its receiving antennas.
2.Transmitting device is TP link
3. I am using Dell inspiron N5110 for receving and processing CSI with intel 5300 wireless card.

One more issue is: I have noticed that data is updated in csi.dat file after each 21 seconds and it is updated with increment of 4.1kB. Why it is so? can I change this delay somehow?

dhalperi, 2015-07-02:
- If one antenna is significantly weaker than others, it may or may not be used often.
- You can have issues if Bluetooth hardware is present in the laptop, it may turn on and off some antennas.
- Dell may have not connected all the antennas, and even if connected they may be 2.4 GHz only antennas.
- There may be something funky with the interaction with the TP link.

As for the way that csi.dat is updated:
- In `log_to_file`, we write each message to the file as it comes in. However, that really just puts the bytes in a buffer, it does not force them to be written to disk. Linux chooses when to write to disk.

- 4 kiB, or 4096 bytes (~4.1kB) is a typical page size, so it would surprise me if the default behavior is to force the writing to happen every time the buffer crosses 4096 bytes. This would explain the behavior you're seeing.

- you can force `log_to_file` to write to disk by inserting a call to `fflush` [here](https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/netlink/log_to_file.c#L91). It would look like this:

    ```c
    ...
    ret = fwrite(cmsg->data, 1, l, out);
    fflush(out);
    if (count % 100 == 0)
    ...
    ```

hmlsi, 2015-07-06:
I am using fflush(out); but csi.dat file is still being updated after 4.1kB

dhalperi, 2015-07-06:
http://stackoverflow.com/questions/2340610/difference-between-fflush-and-fsync

closed 2015-07-06 by dhalperi

hmlsi, 2015-07-06:
now, I can understand what actually fflush is meant to do. But, the situation is still same. Can you please explain?

dhalperi, 2015-07-06:
use fsync in addition to fflush if you want the changes to be observable on
disk. However, you should be aware that, depending on the type of storage
you use, you will be adding hundreds of milliseconds to seconds to your
call.

If you really want real-time CSI, you should just listen to the socket used
by the kernel connector module and process the messages that way. You don't
need to go through the log_to_file and disk.

hmlsi, 2015-07-07:
Ok. Thanks a lot