dhalperi/linux-80211n-csitool-supplementary #72: the received CSI values in a single subcarrier can fluctuate significantly? Is It normal?

xuekevin, 2015-07-03:
I pick some samples in one location, I I draw the plot of one subcarrier's CSI value changes. As you can see. It changes significantly? @linkin8834 
 the x axis it the sample numbers,  I picked up about 120 packets. the y axis is the  module value of CSI message. I got these values by ping method.
[image]

what about your experiments? Is it similar with mine? if not, what tips you have followed? thanks a lot!

linkin8834, 2015-07-03:
Hi Kevin,
   I take a look at mine. It seems to have much smaller fluctuations. My configuration is as follows: wireless router using two spatial streams, laptop using three receiving antennas. The ping interval is 0.01 sec. Is the access point always using fixed number of transmitting streams? My observation is that it sometimes change its number of spatial streams due to rate selection algorithm. On the other hand, Did you try injection and monitoring mode? That one is more stable.

xuekevin, 2015-07-03:
@linkin8834 ，thanks a lot !  my AP also use two spatial streams and my laptop use three receiving antennas.  Most of the time I get 2*3*30 CSI packets, sometimes I can get 1*3*30 packets. the distance between laptop and AP is among 3-10 meters. our environment is lab room which has a lot 
computers, I put the laptop on a chair to pick up samples. all these has 
another questions is sometimes I can get CSI packets without ping, I use wireshark to capture the packets, some are ARP packets. have you met this condition? 
what's more, Have you tried to get CSI packets without ping, such as  get wifi beacons from ap?

linkin8834, 2015-07-03:
@xuekevin No I did not try wireshark. If I use a laptop with hostapd as AP, and connect another laptop to it, I can pick up packages at the AP side when the client tries to associate with it. But, for wireless router as AP, I did not see any package coming in without **ping** it. I don't know if I can get the packages by beacon only. Most papers just said, 'beaconing is about 10Hz, so, we let the transmitter sends packages with 10 packages/sec (in injection mode).'. I guess that they are not actually use the beaconing to get the CSIs.

xuekevin, 2015-07-06:
@linkin8834 ，thanks a lot! I think I need to do some work to make sure there are no packages coming without ping. and when I ping, I did not set ping interval as 0.01 sec. I don't know if it matters. let me try.
you answer help me a lot!

xuekevin, 2015-07-06:
@linkin8834 I want to ask more about your wireless router, Have you made some setting?  the router I use  is TP-Link, I don't connect it with internet, I make some setting as follows.  Have you setup static ARP table? 
[image]

I want to know what's wrong with my experients. hope you are a Chinese.

linkin8834, 2015-07-06:
Hi Kevin,

    I did not setup the arp table. I set the static ip address as
192.168.1.1. Your settings are OKAY.

linkin8834, 2015-07-06:
> <javascript:_e(%7B%7D,'cvml','E-mail:linkin8834@gmail.com');> or
> cc8834@umd.edu <javascript:_e(%7B%7D,'cvml','cc8834@umd.edu');>

Hi Kevin,
   I suggest you try the injection and monitor mode with two laptops. That
should be more stable. Also, I choose channel 13 in the 2.4GHz for the
router.

dpward, 2015-07-06:
@linkin8834 @xuekevin Packets do not have to be generated with the `ping` command. You can use other network control traffic, such as ARP or IPv6 Neighbor Discovery, or application traffic. The interval does not matter.

linkin8834, 2015-07-06:
Hi David,
    Thanks for your suggestion. I was thinking that how fast you ping the AP would make it select a high data rate to turn on the 802.1n transmission...I have heard iperf a lot but have never tried that yet.

dpward, 2015-07-06:
That's true, you could influence the rate control algorithm on a commercial AP. I would expect that along with the interval, increasing the size of the packets (with ping, using the `-s` option) would also have an effect.

linkin8834, 2015-07-06:
Hi David,
     Thanks.

xuekevin, 2015-07-07:
thanks a lot! @linkin8834  @dpward

closed 2015-07-07 by dhalperi