dhalperi/linux-80211n-csitool-supplementary #73: some questions about CSI data

ozawali, 2015-07-06:
Hello,
I have question about the CSI collected in a 2X3 links. There is D-Link AP(commercial AP) as the  transmitting device which has two antennas and Desktop computer with intel 5300 wireless card as receiving device which has three antennas. 
When processing the CSI with matlab, the CSI plot for one channel which transmits about 100 packets looks like this.
[image]
I did the test in a typical office room. There isn’t any movement or change in the room and I get about 100 packets. This figure just describes one link about csi(1,1,:) at extended period, I can’t understand why there are two kinds of waveforms in one channel and there is stable waveform in theory.

dhalperi, 2015-07-06:
This looks like the transmitter is using two different transmission modes, likely as part of rate selection. Either 1-stream vs 2-stream, 1-stream antenna selection, antenna selection + beamforming, etc.

ozawali, 2015-07-06:
@dhalperi  this phenomenon that the picture show, sometimes the line go up then the line go down. you mean that the transmission rate change can aslo change the CSI data?  can you give me some advices than I can check the question. Thank you very much!

dhalperi, 2015-07-06:
The transmitter can change all sorts of variables. It may use antenna
selection, which results in completely different csi of course. It may also
apply spatial mapping when switching from 1 to 2 streams, so there is no
guarantee that 1x1 csi looks like 2x1 csi.

There is simply no way to do controlled experiments with commercial aps.
The only way to do it properly is to use the Intel 5300 at both sides and
use injection or ad hoc mode for full control.

ozawali, 2015-07-07:
[image]
Hi,I did the test again, this time I changed the AP type(TP-Link commercial AP),this AP also has two antennas. the result seems like nomal,those lines are four of the 2X3 Links

so I think that the problem is associated with the AP. The first time I used the AP which brand is D-Link 2310,the second time I used the TP-Link. the result of the second time seems right. I was very interested in this phenomenon.   can you recommend some materials releated to this phenomenon that I can figure out those knowledge. Thank you very much!

dhalperi, 2015-07-07:
A good reference on all the things you have to know when modeling an 802.11n link is, if I don't say so myself, Chapter 4 of my Ph.D. thesis :) https://github.com/dhalperi/thesis/raw/master/document/dhalperi_thesis.pdf

ozawali, 2015-07-07:
@dhalperi  Thank you very much!

hmlsi, 2015-07-08:
[image]
[image]
I hope this issue is related to me thats why posting question here. 
Each graph is showing magnitude plot of 12 received packets. I am not sure about how it should look like. Why there is dip at start and end? is this behavior is normal? I am using TP link as AP having one antenna
Thanks

dhalperi, 2015-07-08:
Yes, the dip on the edge subcarriers is always there. It's likely an effect
of the hardware pulse shaping and or filtering.

hmlsi, 2015-07-08:
Thanks for quick response as always

closed 2015-07-08 by dhalperi

soltanaghaei, 2016-05-26:
@dhalperi @ozawali @hmlsi @ippeb @dpward I have exactly the same problem, but in injection mode. When I changed the channel number from 64 to 10, I got the following graph:
[image]

The room was empty during the test and the two computers with 5300 card worked on injection mode in channel number 10 HT20 and 6mbps. In addition, both of the laptops have 2 connected antennas and the final calculated CSI matrix has the size of 1x2x30

I would be grateful if you can help me realize what the issue is.