dhalperi/linux-80211n-csitool-supplementary #74: PC in AP mode can't work well

co12, 2015-07-10:
Hi, I configure a PC with IWL 5300 to be the AP mode using hostapd and dhcp-server, but when it starts and one station connects to it, after some time other PC or smart phone can't scan the AP, the AP is invisible. Moreover, when I disconnect to the AP, the station can't no longer connect to AP. I googled some days for the problem, but still not solved, please help! thank you

co12, 2015-07-10:
What is more, only one station can connect to the AP, and after a period of time it can automatically disconnect, and never connect again

dhalperi, 2015-07-10:
Please read the FAQ, especially #10

co12, 2015-07-10:
I can't download the LORCONv1 from http://802.11ninja.net/svn/lorcon/branch/lorcon-old/
because it becomes to http://802.11ninja.net/?f, and nothing is there.

Can you send me the file?

dhalperi, 2015-07-10:
Sorry, I cannot help with problems from external servers going down.

dhalperi, 2015-07-16:
Hi @co12 .

Would you be willing to try the injection setup again, but instead use the mirror of `lorcon-old` I put on github? https://github.com/dhalperi/lorcon-old

co12, 2015-07-16:
Hi @dhalperi 
I find the mirror of lorcon-old is sent from @linkin8834, I also copy from him and it works well. So I believe the mirror file is correct without problem

closed 2015-08-10 by dhalperi

linkin8834, 2015-08-20:
@co12  I tried the master mode with 5300 before, and it ended up like what you said: the device that you conect to the 5300 AP would be dropped soon and the AP would literally die :( That should be a issue with the master mode defects on the 5300 chip itself, as mentioned in the FAQ.